PrintMakerLab/mks-wifi-plugin #1: Installation guide

svenddk, 2019-12-10:
Hi,

Just an idea: Would it be possible that you add a short installation guide? Because I am not familiar with the Cura folder structure so it would be a great help ... and today I did not manage to make the plug in work. I just managed to crash Cura 4.4a couple of times.

BR svenddk

svenddk, 2019-12-10:
BTW, a great initiative that you fixed the plugin for Cura 4.4 ... seems that Makerbase don't care so much.
Thanks!

svenddk, 2019-12-12:
Solved. I found out the Cura folder structure.

Jeredian, 2019-12-12:
> Solved. I found out the Cura folder structure.

Sorry for the long answering. 
I thought about placing plugin into Cura Marketplace to easy install, but still not found information on how to do that.

I'll add short help steps to install plugin in the next commit.
About Cura crash, you can look at the crash report. If it looks like the screenshot in this issue [Issue2](https://github.com/Jeredian/mks-wifi-plugin/issues/2)  , so I'm working on it right now

svenddk, 2019-12-12:
No problem, it's working without issues. Thanks.

BR svenddk

svenddk, 2019-12-12:
You can access the Cura Marketplace here http://contribute.ultimaker.com/ then create an account. Log in to your account and click on the squares in the upper right corner - select "Contributor portal". New pop-up asks you to create a contributor account, from there you should be able to share your plugin. It asks for your website, for this I used my github page. Seems to work!
BR svenddk

Jeredian, 2019-12-12:
> You can access the Cura Marketplace here http://contribute.ultimaker.com/ then create an account. Log in to your account and click on the squares in the upper right corner - select "Contributor [...]

Thank you for helping. I'll try to add this plugin to Marketplace.

Jeredian, 2019-12-12:
I submitted a plugin to Marketplace, let's wait when it be approved

Jeredian, 2019-12-17:
The plugin now available at Cura MarketPlace

closed 2019-12-17 by Jeredian