PrintMakerLab/mks-wifi-plugin #8: MKS plugin for Cura 4.1

bilou24an199, 2019-12-21:
hello i found trouble with the plugin for Cura 4.1

i run printing on the sbase 1.3 with the TFT 35. Printing from USB stick

when i want to make a gcode for tft = if a post processing is done in Cura 4.1 to had a pause at Z height the picture of the part in tft is not good read by the tft and the printing not loading. no action on heat bed and extruder

i think that i have trouble too, to have a real "pause during the printing = M0 code is not recognized and M600 not working properly = not a real pause = only a quick stop but no action to run again the printing, the print continue it seems that the M600 is not correctly read by the tft

Jeredian, 2019-12-21:
Hello, 

If I understand right, you want to pause printing via gcode. In this way it's related to Cura gcode processing, and not with wifi plugin.
After Wifi plugin uploaded file to sdcard\usb stick, it sends M23 and M24 command to print file.
If you want to pause/resume print from Monitor in Cura,  you can use Resume\Pause button. It send M25 command to pause and M24 command to continue printing.

About model preview on the display, you can get latest from master branch with changes for it.

bilou24an199, 2019-12-21:
Hello,

sorry for my question which is not in the good section but i have found a link on the "official" MKS Github
https://github.com/makerbase-mks/Software/issues/14

and this link was about the mks tft plugin, so you are right i am not in the good section but thank you very much for your support and your reactivity to do an anwser because on th official one, it's usually to have no support from the MKS team.

thanks a lot for your support.

but i think really that  there is a trouble to generate the tft gcode file with a post processing option activated in cura 4.1

Jeredian, 2019-12-21:
There is another thing, you can save the file not like "save to tft file", but as "save to file". and then manually copy it to usb stick. if there will be the same issue, this thing will be related to Cura slice process.
Also, you can try to update Cura version to the latest (4.4.0+), and check is it solve your issue.

bilou24an199, 2019-12-21:
thank for the repply

i tried to copy the begin of the tft gcode (lines with M10086 beginning) in the gocode with the post processing but i don't why the graphic of the file is totally corrupt.

i'll follow your advice and try to pass to the last version of cura to check is the pb is the same or not.

for the pause in tft and Mks board i think there is a big trouble

https://github.com/makerbase-mks/MKS-TFT/issues/80

no command is correct = M0 / M25 / M600

the issue for the moment is to do a pause with the G4 command and to put a timer which is not usefull = necessary to be near the printer to insert the magnet to the cavity (it is my goal) becaus if the timer is out the cavity is empty

i will prefer to use a real pause function and to push a run button on the TFT to continue the print

Jeredian, 2019-12-21:
I compared code with saving "to TFT file" with common "to file". They looked like similar, except saving settings at the end of the file, if they exist. Not sure is an issue related to it or not, but I'll check this tomorrow. 
But if this trouble with the firmware I'm not sure how can I help.

Jeredian, 2020-01-06:
I'm closing issue because it's not related to the plugin

closed 2020-01-06 by Jeredian