PrintMakerLab/mks-wifi-plugin #11: The previous model review in Cyra 4.4.1 does not work correctly.

Error26, 2020-01-13:
The previous model review does not work correctly. 
I have MKS TFT35 v1.0 firmware 1.0.2.
Cura version 4.4.1 Plugin version 1.0.2
[image]
[image]

Error26, 2020-01-13:
I can give an old version of Cura with a working plugin.
https://drive.google.com/open?id=19DTEfol_n55qEbuVwAuwZxx4jwVb4mWJ

Jeredian, 2020-01-13:
I seems that your preview image size is different than mine.  Because we have different firmware.
I can suggest some simple steps to find your preview size. And I will add it in future releases of the plugin (something like select box for preview size in the plugin popup).
So here are the steps:
1. Close Cura.
2. Go to Plugin folder.
Open the Cura configuration folder. It can be found via Help -> Show Configuration Folder inside Cura.  After that go to "plugins -> MKSWifiPlugin -> MKSWifiPlugin" folder.
3. Find file "SaveOutputDevice.py" and open it in notepad, or code line editor (like VSCode)
4. Find the line with code `save_file.write(utils.add_screenshot(image, 50, 50, ";simage:"))` . It's near line 120.
5. Change the size to a little bit bigger (f.ex. like 100, or 75) for "simage". Pair of numbers should be the same (75, 75 or 100, 100)
6. Save file changes
7. Run Cura, and send the sliced model to the printer again.

You can play with size, find acceptable for your firmware. When a new preview will look fine let me know, about the size of the image.
It's a little workaround until the next plugin release.

Jeredian, 2020-01-13:
If the older version of the plugin works for you, there will be size like this
`save_file.write(utils.add_screenshot(image, 100, 100, ";simage:"))`.

sizanjavad, 2020-01-22:
> I seems that your preview image size is different than mine. Because we have different firmware.
> I can suggest some simple steps to find your preview size. And I will add it in future releases of [...]

worked for me. thank you.

Operrr, 2020-01-30:
Many thanks to the author. Everything is fine.

Bradley-Liu, 2020-05-10:
Cura 4.6 not working I am using 1.0.5 version tft35 preview not working

Elaugaste, 2020-05-14:
Привет, извините что по русски, мой инглиш слишком плох. 

Было бы крайне здорово сделать формирование превью опциональным, т.к. существуют принтеры которые не умеют адекватно переваривать лишний хлам в файле с гкодом. 

MKS Robin mini например внезапно может забурится соплом в стол или пытаться унести голову за аппаратные пределы

Jeredian, 2020-05-15:
@Bradley-Liu , new Cura has a little bit different API. I'll update the plugin on next week to support the new version, and also with fixes for preview.

Jeredian, 2020-05-15:
@Elaugaste , Добрый вечер. Рад что у Вас получилось разобраться до того нового релиза.
Если не сложно то сделайте пулл реквест в мой репозиторий, и в выходные, либо на следующей неделе он будет доступен через cura marketplace

Elaugaste, 2020-05-15:
@Jeredian 
пулл реквест сделал, но я отфорковался не от вашего мастера а от другого форка.
потому там только последний коммит содержит нужные изменения...

karabas2011, 2020-05-28:
плагин будет коннектиться к произвольному принтеру с MKS TFT35 + MKS wifi?
Cura 4.6.1 не коннектится и не видит номер  прошивки

Jeredian, 2020-05-28:
@karabas2011 , по идее должен.  Более развернуто могут ответить разработчики makerbase я думаю. либо спроси в группе https://t.me/K_3_D

Номер прошивки не видит потому что makerBase не прикрутили это. в ближайших релизах возможно сами допилим.

Jeredian, 2020-05-28:
@karabas2011 , сама по себе Cura не умеет соединяться с принтером , а плагин может только к тем кого видит в своей подсети, конечно с учетом что это MKS TFT35,  и в идеале плата mks robin

karabas2011, 2020-05-28:
ну да плагин я поставил, зайти на вайфайку могу броузером или пронтерфейсом, то есть она доступна по сети.

karabas2011, 2020-05-28:
кстати если коннектиться пронтерфейсом половина команд не отзывается. например M119, M503  и тд. а M105 отзвается. M115 выводит строку самой вайфайки. Интересно есть способ насквозь команды пропускать в принтер?

Jeredian, 2020-05-28:
@karabas2011 , может у тебя плата какая специфичная ?
вроде как там просто команды должны посылаться на плату, и обратный ответ от нее

Jeredian, 2020-05-28:
@karabas2011 ,  касательно команду думаю надо создавать issue тут : https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT3.5

Jeredian, 2020-05-28:
@karabas2011 , про подключение через Cura - распиши как ты это делал, может какой то шаг пропустил. 
Если что авто поиск принтера в сети пока еще не добавлен, там надо в ручную ip принтера указывать. и потом нажимать connect button

karabas2011, 2020-05-28:
так а там собственно нечего делать. как только плагин поставил он тут же вписался в текущий принтер как кнопка MKS WIFI. Нажал - ввёл IP  а дальше фиг.

Elaugaste, 2020-05-28:
логи куры нужно смотреть

karabas2011, 2020-05-28:
таймаут и реконнекты

karabas2011, 2020-05-28:
подробностей не пишет.

Elaugaste, 2020-05-28:
@karabas2011 попробуйте прожать telnet на порт 8080.  Плагин стучится в него,  возможно у вас на другой порт нужно ходить

adripo, 2020-05-29:
I have a similar error 

> I seems that your preview image size is different than mine. Because we have different firmware.
> I can suggest some simple steps to find your preview size. And I will add it in future releases of [...]

I have a similar error, but this didn't help.

My image is translated and have strange colors.
Have a look
[image]

Jeredian, 2020-05-29:
Hi, @adripo , you can try to use different values "50, 50",  "75, 75",  "100, 100".
At next releases, I hope, I will add select box for that size

adripo, 2020-05-29:
I tried 25, 50, 75, 100, 200 but nothing changed. Every time I diconeected, closed cura, changed value, reopened cura, connected, slice and send.
Do you have any other idea?
Thanks

Jeredian, 2020-05-29:
There is 2 places with image sizing:
1. SaveOutputDevice.py  - line 120
2. MKSOutputDevice.py - line 619

As first step it's to set both of them for simage - as 100
the second one try to comment both lines and check will image disappear or not

That's a very simple bug, but it's hard to implement selector for different sizes inside QML file, without project .pro file.. because qt designer can't import uranium framework and cura engine to preview form
If this two steps didn't help, let me know I'll try to speed up work on this task.

adripo, 2020-05-30:
Thank you for the fast reply!
I changed the value (to 50) also in MKSOutputDevice and now the picture is perfect.
Thanks for the help.
Can you explain me better how is this programmed? Maybe I can help you. I had a look at qt designer. How does this interface with cura? Can you share any documentation with me please?

Jeredian, 2020-05-31:
In short, it makes an image with selected size and puts it at the start of gcode file. When printer will open gcode file, the image will be treated and you can saw it at the screen.
About UI, it's all qml files inside the project. MKSConnectBtn.qml - seems to be UI where we can place select box with sizes (small, medium, large), these settings can be saved as we used in the printer IP list.
Also, you can look at Cura wiki, to ger information about the environment and references.
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
https://github.com/Ultimaker/Cura/wiki/QML-Best-Practices

closed 2021-02-07 by Jeredian

Volt-AmperAI, 2021-04-13:
Hello everyone, in Cura version 4.8 there is no line save_file.write (utils.add_screenshot (image, 50, 50, "; simage:")) in the SaveOutputDevice.py file. Perhaps the new version of Cura does not support the preview window.

Jeredian, 2021-04-13:
Hi @Volt-AmperAI , SaveOutputDevice.py - it's not Cura, it's our plugin code.
Since the latest release, we moved model preview to the plugin settings window. Please check it out

Volt-AmperAI, 2021-04-13:
MKSWIFI plugins version
[URL=https://fastpic.ru/view/114/2021/0413/b6dee9550655f345775ca44f1b969e73.png.html][IMG]https://i114.fastpic.ru/thumb/2021/0413/73/b6dee9550655f345775ca44f1b969e73.jpeg[/IMG][/URL]
The problem is that the image goes beyond the boundaries of the TFT35 v1.0 screen. And for some reason, artifacts on the screen. This is if you save to TFT in Cura.
[URL=https://fastpic.ru/view/114/2021/0413/462395e0346eca38c561d737899572f3.jpeg.html][IMG]https://i114.fastpic.ru/thumb/2021/0413/f3/462395e0346eca38c561d737899572f3.jpeg[/IMG][/URL]
[URL=https://fastpic.ru/view/114/2021/0413/f157f05916702a8a5afbdc21862c9ec5.jpeg.html][IMG]https://i114.fastpic.ru/thumb/2021/0413/c5/f157f05916702a8a5afbdc21862c9ec5.jpeg[/IMG][/URL]

Jeredian, 2021-04-13:
@Volt-AmperAI , you should select the correct screenshot settings. There are several of them (base, none, and for some printer models). If none of them will be correct for you, you can create an issue for it, and if you know the correct size values, you also can make a PR. 
[image]
[image]

Volt-AmperAI, 2021-04-13:
my printer does not go online at all, the MKSWIFI module is working, the phone sees it, but for some reason the plugin does not work in Cura, maybe the MKSWIFI plugin does not work in Cura 4.8? Maybe someone knows?
[image]

Jeredian, 2021-04-13:
@Volt-AmperAI , 
1. Remove IP of your current printer
2. Restart the Cura
3. Add new IP and press connect button.

FYI, for that type of questions, I suggest to you ask it at telegram channel https://t.me/mks_wifi_plugin_dev

Volt-AmperAI, 2021-04-13:
Does not connect to the network, does not find a specific IP address. Although the IP address is indicated on the phone, the same as I enter in Cura
[image]

PR #106 (2021-02-07): MWP-11 Model image fix
closes #11
closes #45
closes #80