PrintMakerLab/mks-wifi-plugin #12: File transfer by wi-fi to MKS-TFT35 broken in Cura 4.5.0

svenddk, 2020-03-04:
Hi,

File transfer by wi-fi to MKS-TFT35 is not working. When file has been transferred then printer does not start heating, it only starts printing in the air and then ends the print job within a few seconds. Same failure when saving tft file to sd card and then putting sd card into TFT35. Print preview pictue on TFT35 is blurred. Problem for TFT35 firmware 1.0.3 and 1.0.4 so I guess that there is a mismatch between Cura 4.5.0 and the mks-tft-plugin. No problems when storing G-code file from Cura to sd card and then putting sd card into TFT35.

I hope you can fix it because the mks-tft-plugin is great !

BR Svend

Bradley-Liu, 2020-03-05:
cura 4.4 has the same issue.

svenddk, 2020-03-05:
For me, the jeredian mks-wifi-plugin worked perfectly in Cura 4.4. Problem only with Cura 4.5.

MKS plugins https://github.com/makerbase-mks/Software/tree/master/MKS%20Plugin have not been updated for a long time. This caused problems with Cura 4.4. There are many plugins, the  jeredian mks-wifi-plugin is one of the good plugins.

Jeredian, 2020-03-10:
After I update Cura to the latest version, I'll check this and hopefully can fix it

svenddk, 2020-03-10:
Perfect, thanks. I trust that youcan fix it and look forward to the update.

makerbase-mks, 2020-04-16:
@svenddk You can try this version: https://github.com/makerbase-mks/mks-wifi-plugin

svenddk, 2020-04-19:
Please release a working version in Cura marketplace, thanks. There has been way too much trouble with Makerbase wifi plugin for Cura due to lack of support and updates from Makerbase so I am looking for another solution.

Jeredian, 2020-04-19:
@svenddk , can you test a version from the pull request #13 ? 
If it solves your issue please let me know?
Does it happen with all models that you tried to print, or only with a one?
Sorry for a long absence, I was a little busy on the main work but now I'm back
On next week I'll merge the latest changes from makerbase, and then will start work on your issue.

Jeredian, 2020-04-19:
@svenddk , Also can you check your issue, on the latest beta version of Cura?
(https://github.com/Ultimaker/Cura/releases/tag/4.6-beta)
Because if it happens only since 4.5 maybe they change something Cura API, and when we slice model we have a gap between surface and model.
Also please take a look at your start G-code for the printer, this code always copy-pasting to the start of wifi-sended g-code, and maybe we have an issue in that place.
Can you send me a sample of G-code files, one with wifi-saved and the other when you save manually to sd-card?  I'll compare them, to found what happens inside Cura and plugin.

svenddk, 2020-04-19:
@Jeredian sorry but no, I have given up the Makerbase wifi plugin

closed 2020-04-19 by svenddk

MKS-Sean, 2020-05-05:
https://github.com/makerbase-mks/mks-wifi-plugin
I tested this plug-in. TFT35 transfers files via wifi and starts printing. It can print normally. The print interface cannot display the preview.The preview can be displayed normally if you load the file from the SD card and start printing.The problem was improper handling of TFT35 and cura plugin.

Jeredian, 2020-05-05:
Hi @MKS-Sean ,  thanks for a testing version from makerbase. I'm glad that printing is working for you. 
About image preview size - it's a known issue, and the next release at the market place will fix that behavior. 
Also, I described this issue and short workaround how you can fix preview image size  https://github.com/Jeredian/mks-wifi-plugin/issues/11