PrintMakerLab/mks-wifi-plugin #18: File name already exists on sd-card

adripo, 2020-05-29:
Hi, if I click yes on the prompt to change the name because it is too long and there is another file with the same name as the one that I inserted I receive an error

```Traceback (most recent call last):
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\MKSWifiPlugin\MKSOutputDevice.py", line 803, in recheckfilename
    yesbtn.clicked.connect(self._startPrint(filename))
TypeError: argument 1 has unexpected type 'NoneType'
```

<details>
<summary>Logs</summary>

```
Thread 0x00004c28 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000037b8 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002fd4 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000396c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004280 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000025c8 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f80 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001bc4 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000038d0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002c60 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\MKSWifiPlugin\MKSOutputDevicePlugin.py", line 277 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003078 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004968 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002460 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001e04 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000096c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000e40 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033dc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000335c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000045cc (most recent call first):
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 379 in _logInfoWidget
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 180 in _createDialog
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.6-exe\build\inst\bin\cura_app.py", line 163 in exceptHook
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\MKSWifiPlugin\MKSOutputDevice.py", line 765 in startPrint
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\MKSWifiPlugin\MKSOutputDevice.py", line 686 in requestWrite
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 141 in _writeToDevice
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\Event.py", line 218 in call
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 464 in event
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1013 in event
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 813 in run
  File "X:\4.6-exe\build\inst\bin\cura_app.py", line 231 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```
</details>

Can you help me?

Jeredian, 2020-05-29:
Hi, @adripo , I'll try to reproduce this issue, and then back to you.

Jeredian, 2020-05-29:
Btw, you can try to get latest from master branch, because if I remember right there was some fixes from makerbase team.

adripo, 2020-05-29:
I ahve the last version from your repo, master branch

adripo, 2020-05-29:
Let me check if there is any update.

adripo, 2020-05-29:
I have the latest release. For the latest code from their master branch https://github.com/Ultimaker/Cura I need to build it manually and I cannot do it right now

Jeredian, 2020-05-29:
ok, I'll check that

Elkin-Vasily, 2021-01-17:
Is this issue actual for the latest commit of this repo?

closed 2021-01-20 by Jeredian

adripo, 2021-03-22:
I confirm, the issue is solved. Thanks

PR #66 (2021-01-20): TypeError: argument 1 has unexpected type 'NoneType' error fix
resolves #18