PrintMakerLab/mks-wifi-plugin #19: Cura interface layout overlap

adripo, 2020-05-30:
Sorry for opening another bug, but I want to keep thing separated.

I have a bug in the layout of the plugin on cura 4.4.1
Here's a screenshot:
<details>

[image]
</details>

Some of the buttons are overlapping in the Printer control and Active print sections.
Thanks

ravetam, 2020-06-26:
MonitorItem4x.qml

Changing the y from 450 to 600 seems to fix the issue

```
Row
            {
                x:10
                y:600
                spacing: 10
                anchors
                {
                    leftMargin: UM.Theme.getSize("default_margin").width
                    rightMargin: UM.Theme.getSize("default_margin").width
                    horizontalCenter: parent.horizontalCenter
                    // bottomMargin: UM.Theme.getSize("default_margin").height
                }
                Button
                {
                    id: homebutton2
                    height: UM.Theme.getSize("save_button_save_to_button").height
                    // text: catalog.i18nc("@label", "Fan ON");
                    text: currentLanguage == "zh_CN" ? "打开风扇" : "Open Fan"
                    onClicked: Cura.MachineManager.printerOutputDevices[0].openfan()
```

PigeonFX, 2020-07-01:
> MonitorItem4x.qml
>
> Changing the y from 450 to 600 seems to fix the issue
>
> ```
> Row
>             {
>                 x:10
>                 y:600
>                 spacing: 10 [...]

This doesn't fix the 2 Extrude buttons being ontop of the Z-axis buttons.

PigeonFX, 2020-07-01:
Seems change the following lines in MonitorItem4x.qml fixes the extruder buttons:

504: x:310
505: y:243

closed 2021-01-11 by Jeredian