PrintMakerLab/mks-wifi-plugin #20: Plugin doesn't work via wifi with multiple istances of cura

adripo, 2020-05-30:
I tried to open different istances of cura with different objects to print, but only the first instance is connected via wifi with the printer. All the others doesn't.
If I go to MKS WIFI and try to connect manually via wifi nothing happens. The "connect" button remains clickable, but there is no message.
After that, if I close the working instance, none will work anymore. Even if I close them all and open just one window of cura. I have to open and close cura multiple times until it starts working again.
Is there any way to see a log? Or even a simple error message?
Thanks

Elkin-Vasily, 2021-01-17:
Look at Cura log file, all plugins are logging their activities there

Elkin-Vasily, 2021-11-21:
MKS WIFI module supports only one connection at a time.

closed 2021-11-21 by Elkin-Vasily