PrintMakerLab/mks-wifi-plugin #25: Plugin when installed via the Marketplace doesn't work.

PigeonFX, 2020-07-01:
If I install via the Marketplace it can't find my printer via IP, but if I manually install the plug-in it works fine (apart from the monitor layout being jacked up). 

Using PC and Cura 4.6.1 with a Makerbase MKS Robin WIFI ESP8266 WIFI module

Jeredian, 2020-07-27:
Hi, are you tried the last version from the master branch?
I not updated plugin for Cura 4.6.* because they have too many bugs in the new version, and also changes at API version, so maybe it related to that thing. I planned to update the plugin at Marketplace at the start of August.

dgorbunov, 2020-07-28:
This happens to me as well, plugin from Marketplace doesn't work.

Emphantom4d, 2020-07-30:
Hey . I also installed it from the market, it does not work. install from github, does not work either.
what could be?
cura 4.6.2

Jeredian, 2020-09-07:
Hi, can you try to use the latest version from the master branch?
I hope I'll update it on Market Place at this weekend, but if someones want to test before it, it will be helpful.

closed 2021-01-17 by Jeredian