PrintMakerLab/mks-wifi-plugin #28: can't access sd card from wif plugin

Bradley-Liu, 2020-07-27:
i am using cura 4.3 and your wif plugin. recently i've updated the firmware of my tft35 to v1.07. after that , i just can't access my sd card file using the plugin anymore. if i want to print files that are already in the sd card, i have to transmite the file again. please help. Thank you.

Jeredian, 2020-07-27:
Hi, if some feature doesn't work after you upgrade firmware it's related to new firmware code and it methods support. I think it will be better if you ask the same question to MKS developers, because it is something on their side.
As you can see inside source code, plugin connecting to firmware and sending get/post requests, so it doesn't interact with sd card directly.

Elkin-Vasily, 2021-01-18:
Is this issue actual with lates commit from this repo?

Elkin-Vasily, 2021-02-06:
Looks like this issue is obsolete

closed 2021-02-06 by Elkin-Vasily