PrintMakerLab/mks-wifi-plugin #32: Printer ramms nuzzle into print after print

berberic2, 2020-08-29:
Hi,

My FLSUN-QQ-S has rammed its nozzle into the print after the print is finished several times.
The prints were started via mks-wifi-plugin from cura (4.7.0) and finished fine. At the end the printer did a home and then rammed the nuzzle straight down in the finished print.

Jeredian, 2020-08-31:
Hi, you should look at ending GCode, in Cura Printer Settings.
It's not related to the plugin, because it just sends sliced model with start/end code settings to sd card

berberic2, 2020-09-02:
The gcode probably is fine. If it weren’t this would happen every time.

Jeredian, 2020-09-03:
Can you try the same printing at another version of Cura, 4.6.* ? If it will be ok, it's related to slicing code at Cura.
And, anyway please check gcode of a file that was with bad printing, there is should be lines like that

> G28 X0 Y0  ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning

If Cura does not add this, or for some way changed the position of the head it can go in another way. 
Or maybe M84 command can be the result that the head moves to the downside printer. But this settings is for the end code of Printer params.

The plugin just sends gcode with preview by wifi and check printer temperature.
Preview of the model placed at the start of gcode, so it can't be related to the printing process.

You can also try to replace/format sd card, maybe (not sure) it has wrong sectors, and some lines of code can't be read

berberic2, 2020-09-03:
This is the content of a print, where the nozzles is rammed into the object from the SD-Card
__________
--boundary_.oOo._jQoN6tZNpyQ8Cgb5SbMmDQ8JoubrXvN6
Content-Disposition: form-data; name="file"; filename="FQQS_overhang_test.gcode"

;simage:
**_[image-data]_
_[GCode]_**
_M107
M107 T0
M104 S0
M104 S0 T1
M140 S0
G92 E0
G91
G1 E-1 F300 
G1 Z+0.5 E-5 X-20 Y-20 F9000
G28 X0 Y0
M84 ;steppers off
G90 ;absolute positioning_

M82 ;absolute extrusion mode
M104 S0
;End of Gcode

--boundary_.oOo._jQoN6tZNpyQ8Cgb5SbMmDQ8JoubrXvN6--
__________

The slanted/italics part is the end-gcode from the printer-settings. This looks o.k. for me. 
(There is no boundary between image and gcode?)

Maybe the boundary at the end confuse the printer? Or the “check printer temperatures” interaction with the printer (if cura is terminated during print)?

Jeredian, 2020-09-03:
The end code is a little bit different than in the Cura default profile for FLSUN-QQ-S.

> https://github.com/Ultimaker/Cura/blob/master/resources/definitions/flsun_qq_s.def.json

But your end-gcode looks as it should too. So there are no issues at gcode.

Jeredian, 2020-09-03:
Can you try to use the latest version of the plugin from the master branch?
There were some compatibility issues with 4.7.0, at MarketPlace at now placed an older version of the plugin. I planned to update it on this week

Jeredian, 2020-09-03:
> Maybe the boundary at the end confuse the printer?  Or the “check printer temperatures” interaction with the printer (if cura is terminated during print)?

this boundary should not confuse the printer, because the last line that will be read by the printer will be an ';End of Gcode'
about Cura termination during print - this should not be related too, because after you send gcode, it will run only with commands inside file. and any other processed only can pause the print.

Can you try to use option save to file, and put it manually to sd card , because it will be saved only by Cura (without plugin) and if the behavior will be the same - you should ask the question at the Cura repository.

peteristhegreat, 2020-09-17:
I am troubleshooting this same problem, and I am seeing 200+ lines of bitmap data? in the gcode file to the printer (FLSUN QQ-S Pro)

```
--boundary_.oOo._p3CszdJXWeEOOKxcy4+IgoUpB0W0wL6y
Content-Disposition: form-data; name="file"; filename="my_print.gcode"

;simage:00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;;; Repeated lines of image? data
```

before the normal beginning code (normal meaning the sd file when saved as a mounted drive by cura).

```
;FLAVOR:Repetier
;TIME:13611
;MINX:-103.022
;MINY:-117.569
```

And then at the end of the file if I save it directly to the SD card from CURA (mounted drive) it ends like this:

```
G28 X0 Y0
M84 ;steppers off
G90 ;absolute positioning

M82 ;absolute extrusion mode
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Draft #2\\ndefin
;SETTING_3 ition = flsun_qq_s\\n\\n[metadata]\\nquality_type = draft\\nsetting_v
;SETTING_3 ersion = 15\\ntype = quality_changes\\n\\n[values]\\nadhesion_type = 
;SETTING_3 skirt\\nsupport_structure = tree\\nsupport_type = buildplate\\n\\n", 
;SETTING_3 "extruder_quality": ["[general]\\nversion = 4\\nname = Draft #2\\ndef
;SETTING_3 inition = flsun_qq_s\\n\\n[metadata]\\nquality_type = draft\\npositio
;SETTING_3 n = 0\\nsetting_version = 15\\ntype = quality_changes\\n\\n[values]\\
;SETTING_3 ninfill_sparse_density = 20\\nsupport_roof_enable = True\\n\\n"]}

```

If I send the same gcode over this plugin, I get this inside the resulting file on the sd card:

```
G28 X0 Y0
M84 ;steppers off
G90 ;absolute positioning

M82 ;absolute extrusion mode
M104 S0
;End of Gcode

--boundary_.oOo._p3CszdJXWeEOOKxcy4+IgoUpB0W0wL6y--

```

I'll try switching to the master branch and install that version of the plugin; and report back.  This has happened for me for Cura `4.6.2`, `4.7.0`, and `4.7.1` now.

I'm thinking it must be something with the `boundary` line code getting interpreted as Gcode.  And it probably doesn't happen for every send over MKS because only some of the boundary lines have codes that could be rendered by the printer.

Jeredian, 2020-09-19:
Hi @peteristhegreat , can you try the option from plugin "**Save as TFT file**", because in that way the plugin will save all gcode (with image previews) but without direct sending to the printer.  

**boundary**  lines - it's a result after sending data to sd card by post request, and usually they are not used in the printing process.
As we can see at now this issue only happen on FLSUN QQ-S Pro, so maybe the firmware of the printer works with this Gcode in another way. 

Can you attach 3 files of the same model? (sd card saving from Cura, saved as TFT file from plugin and file that was sended by wifi to printer) I will compare them, and maybe we will have something new information about it.

peteristhegreat, 2020-09-19:
Even with the latest code in master, it still did it again for me. 

Here is a smaller print it did it for me just now: 

https://gist.github.com/peteristhegreat/3e2ac8a75295f0472194d5ff744a3e0d

I'm thinking that the Marlin firmware doesn't respect the `; End of Gcode` line properly; though how it parses the extra lines is a mystery to me.

https://marlinfw.org/meta/gcode/

https://reprap.org/wiki/G-code

And here is the main support page for this printer:

https://www.facebook.com/groups/120961628750040/

Jeredian, 2020-09-20:
as we can see in commands we have the same gcode - so you should have the same issue in any type of printing file ( through cura, or through the plugin ). If not there should be some misunderstanding between firmware and plugin. 
What happens if we print the "save as TFT" file? If it will work ok, we have an issue with `--boundary line`, and it's inside the firmware. Because as you can see in the source code, the file was sent to the printer, and we can't control `--boundary` because it's generated by TFT screen firmware.

Another way you can try to remove `M84 ;steppers off`. steppers will be on, and without a command to go down - should be upstairs.

Troutzorz, 2020-09-30:
Hi this was also happening to my FLSUN QQ-S Pro. I edited SaveOutputDevice.py and MKSOutputDevice.py, commenting out 
```
#self._screenShot = utils.take_screenshot()
...
#if self._screenShot:
#    single_string_file_data += utils.add_screenshot(self._screenShot, 100, 100, ";simage:")
#    single_string_file_data += utils.add_screenshot(self._screenShot, 200, 200, ";;gimage:")
#    single_string_file_data += "\r"
```
It has not happened since commenting it out.

Jeredian, 2020-09-30:
@peteristhegreat , @berberic2 , guys can you try to do the same thing?

closed 2020-10-04 by Jeredian

berberic2, 2020-10-05:
> @peteristhegreat , @berberic2 , guys can you try to do the same thing?

No problems with the commented out-version until now (about half a dozen prints).

Jeredian, 2020-10-06:
@berberic2 , that's great.  you can try the latest from the master, there are changes for it. Thanks, @peteristhegreat for the pull request with fixes.

peteristhegreat, 2020-10-26:
FYI: The fix for this is based on the name of the printer in Cura. If you rename the printer, it might still include the screenshot code, and then still ram into the part/bed.  This may cause an issue if you have more than one `FLSUN QQ-S` or you renamed it to `FLSUN QQ-S Pro`, etc.

https://github.com/peteristhegreat/mks-wifi-plugin/blob/306db6c24a6f37dfb967f4a2f3c40ad3a9d2b383/MKSWifiPlugin/utils.py#L32

```
def printer_supports_screenshots(printer_name):
    # printers in denylist sometimes ram the nozzle downwards at the end of a print
    denylist = ["FLSUN QQ-S"]
    return not (printer_name in denylist)
```