PrintMakerLab/mks-wifi-plugin #34: Cura crashes when loading configuration

scmanjarrez, 2020-09-07:
Hi,
Thanks for your plugin.
When I closed cura and tried to open again, Cura crashed, here are the logs:

[image]

<details> <summary> Error traceback </summary>

```
Traceback (most recent call last):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 925, in event
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 411, in event
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189, in call
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 168, in setInitialActiveMachine
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 368, in setActiveMachine
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderManager.py", line 371, in fixSingleExtrusionMachineExtruderDefinition
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 117, in createExtruderStackWithDefaultSetup
AttributeError: 'NoneType' object has no attribute 'getDefaultVariantNode'

```

</details>

<details> <summary> Logs </summary>

```
Thread 0x00000c90 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000eb8 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000d78 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000968 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevicePlugin.py", line 211 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000202c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 78 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002150 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000021ec (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
[... 90 lines cut ...]
  File "X:\4.0\build\inst\bin\cura_app.py", line 110 in exceptHook
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraSplashScreen.py", line 100 in showMessage
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 485 in showSplashMessage
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 183 in startSplashWindowPhase
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 436 in startSplashWindowPhase
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Application.py", line 214 in run
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 287 in run
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 692 in run
  File "X:\4.0\build\inst\bin\cura_app.py", line 132 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

```

</details>

I have installed the plugin from the marketplace.

Jeredian, 2020-09-07:
Hi, can you try to use plugin from the master branch?
If it's not be solved, I'll look and try test on Cura 4.0.

scmanjarrez, 2020-09-10:
Sorry to late response. I have tried multiple times the manual installation without success.

Those are the folder structured tested, none of them worked.

C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSPlugin\MKSPlugin
C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSPlugin\MKSWifiPlugin
C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSWifiPlugin
C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSPlugin
C:\Users\S\AppData\Roaming\cura\4.0\plugins\mks-wifi-plugin-master\MKSPlugin
C:\Users\S\AppData\Roaming\cura\4.0\plugins\mks-wifi-plugin\MKSWifiPlugin

Jeredian, 2020-09-10:
At first, I recommend removing all these folders.
Because if they will be more than one it will produce Cura crash because the name of the plugin the same but the path is different.
After that restart Cura and if it loads without any crashes, install the Wifi plugin from Marketplace.
Restart Cura again, and check if it crashes or not.
If it crashed, download the latest version from the master and replace files. Please, do not create a new folder, just replace already existed files.
F.ex. my folder with plugin files is "cura\4.4\plugins\MKSWifiPlugin\MKSWifiPlugin", you should have the same.

scmanjarrez, 2020-09-10:
I'll do that, thanks.
Note: I just renamed multiple times the same folder, I didn't create multiple copies.

scmanjarrez, 2020-09-10:
I have done what you recommended, but it didn't work.
C:\Users\S\AppData\Roaming\cura\4.0\plugins\MKSWifiPlugin\MKSWifiPlugin

I have installed throught the market (crashes on restart, if I remove the configuration from cura.cfg, it starts without problems), then I replaced all files there with the files present in master branch, but it appears disabled in the installed packages, on restart if I "enable" it, on restart, the plugin stay disabled.

Jeredian, 2020-09-10:
it seems that Cura ver 4.0 save settings of existed printers not properly with "," before IP of the printer.
can you try to use a plugin from MarketPlace, remove these lines at Cura settings file, and setup printer again?
If there will be no "," before ip of printer it should works ok

Jeredian, 2020-09-10:
If it doesn't help I'll try to find version 4.0 and test with it. 
Btw, why you are using this version, not 4.4 or 4.6.2 or the latest 4.7.* ?

Jeredian, 2020-09-10:
Anyway, I'll check this. maybe Cura 4.0 (SDK 6.0.0) and Plugin not fully compatible.

For 4.4.* or 4.6.* - it will works, about the latest version - try the version from the master because **adripo** made some fixes for it.
I hope that I'll update the plugin on this week at MarketPlace because I'm waiting for updates from **MakerBase team**

scmanjarrez, 2020-09-10:
Ok, doing some debugging in the code, I see that first time plugin is instanciated, it creates the variable mkswifi/manual_instances with default value "", then you split it by comma, creating [''], then append the added IP and join by comma, so you have a list with a ["", "192.169.1.49"]. I'm not sure how cura cfg default variables work, maybe is possible to have a default value as empty list, or don't count the empty string when joining the list.

```
mksdebug: manual_instance before split -> 
mksdebug: manual_instance after split -> ['']
mksdebug: manual_instance after append -> ['', '192.168.1.49']
mksdebug: manual_instance after joining list -> ,192.168.1.49
```

scmanjarrez, 2020-09-10:
I have updated Cura to 4.7.1, and it works fine. Thank you very much.

scmanjarrez, 2020-09-11:
Ok, doing a bit of research, I've discovered that isn't a plugin problem but a cura 4.0.0 problem https://github.com/Ultimaker/Cura/issues/5477, so this problem was fixed in later versions.

closed 2020-09-11 by scmanjarrez