PrintMakerLab/mks-wifi-plugin #35: SD card content display problem

Malderin, 2020-09-19:
Hello!
Mks robin nano v.2 board Installed the latest firmware from here: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
Problem with displaying the contents of the memory card. All model previews show the same picture of one of the models.
The plugin is installed from this thread.
[image]

Jeredian, 2020-09-19:
Can you try to restart Cura, add another model, and send it to the printer?
As you can see in the source code of the plugin it can send only the current Cura project preview.

Malderin, 2020-09-19:
Я так понимаю можно писать по русски?
Дело в том что моя плата относительно новая и еще не поддерживает вайфай модуль на программном уровне, не доделали они еще прошивку по нормальному. Я использую плагин мкс вайфай 
только чтобы сохранять файлы с превьюшками и закидывать их на карту памяти вручную.Разумеется все файлы на карте памяти я сохранял через последнюю версию плагина, это не какие то 
старые файлы, все сделано буквально сегодня, чтоб отправить ошибку на гитхаб.
Забыл сказать, во время печати превью моделей отображаются правильно, проблема только при просмотре содержимого карты.

Jeredian, 2020-09-19:
Да можно :) Мне особо разницы от языка нет)
Судя по скриншоту - больше всего шанс что это в прошивке дело, т.к. она использует одно и тоже изображение вместо того чтобы рендерить те что в gcode файле. Хотя сам факт этого довольно странный.
Ты можешь попробовать сохранить файл на карту через save as TFT file, скопировать его в ручную на принтер и проверить - будет ли разница, если не будет то дело точно не в передаче по wifi.
Если знаешь или просто понимаешь немного питон то можешь посмотреть в исходниках на примерно эти строки:
SaveOutputDevice.py - line 120-130
MKSOutputDevice.py - line 910-930
Там происходит добавление файлов в gcode для печати. 
У марлина был какой то косяк с отображением превью.. могу конечно по чату в телеге по искать, но это месяца 3и назад обсуждалось.
Еще можешь спросить в чате K3D в телеге. https://t.me/K_3_D

Malderin, 2020-09-19:
На самом деле я тоже думаю что проблема в прошивке, потомучто на собранной последней прошивке marlin 2.x bagfix все работает нормально. На этой же прошивке от мкс такая вот проблема. Я 
даже больше скажу, если поставить этот плагин из магазина куры, то при заходе через меню на карту памяти получаем зависание интерфейса, помогает только перезагрузка принтера. 
Я им параллельно написал об этом тоже в их ветке, может ответят что.

> Ты можешь попробовать сохранить файл на карту через save as TFT file,
> скопировать его в ручную на принтер и проверить - будет ли разница, если не
> будет то дело точно не в передаче по wifi.
собственно я так и делал. 

> Если знаешь или просто понимаешь немного питон то можешь посмотреть в
> исходниках на примерно эти строки:
увы, не разбираюсь в этом

Jeredian, 2020-09-19:
Как показывает практика гораздо удобнее и надежнее собирать прошивку на чистом марлине самому, либо брать под свой принтер у ребят кто это уже делал. у MakerBase - тех поддержка очень плохая, а прошивка 2.0 с большим количеством багов.

Malderin, 2020-09-19:
Я и собрал сам, но дело в том что в чистом марлине с графической оболочкой, которая появилась сравнительно недавно, отсутствует куча дополнительных пунктов, вот и сел на их прошивку. Нужда 
заставила.

closed 2020-09-19 by Malderin