PrintMakerLab/mks-wifi-plugin #36: Cura freezes while MKS Plugin trying to reconnect to the printer

Mistpal, 2020-09-29:
Printer model Flyingbear Ghost 5
Cura version 4.7.1
MKS Plugin version 1.0.2
OS Windows 10 Pro x64 10.0.19041.508

Hi, I found the reason for the Cura freezing after disconnecting the printer (which was connected via wifi)

0. Cura and Printer were launched. Cura "sees" the printer via wifi
1. Turned off the printer - in the chicken, the printer is supposedly still connected
2. The log Cura% appdata% \ cura \ 4.7 \ cura.log contains "spam" messages from the "MKS Plugin" plugin regarding the wifi connection

in these reconnects and freezes Cura - in the video I drive back and forth with the middle button held down and periodically the chicken freezes.

3. I turn off the printer in Cura via the MKS Plugin plugin

Connect-reconnect logs no longer appear in the Cura log.
Kura no longer freezes.

Looks like the MKS Plugin is the source of the Cura friezes.

Привет, нашел причину зависания Кура после отключения принтера (который был подключен через вифи)
0. Cura и Принтер были запущены. Cura "видит" принтер по wifi
1. Выключил принтер - в куре принтер якобы еще подключен
2. В логе Cura %appdata%\cura\4.7\cura.log идёт "спам" сообщений от плагина "MKS Plugin" по части коннекта по wifi

в эти реконнекты и фризит Cura - на видео с зажатой средней кнопкой вожу туда-сюда и периодически кура фризится.

3. Отключаю в Cura принтер через плагин MKS Plugin

В логе Cura перестают поступать логи по коннект-реконнекту.
Кура больше не фризит.

Похоже плагин MKS Plugin источник фризов Cura.

Mistpal, 2020-09-29:
Video about this report https://www.youtube.com/watch?v=xbZzLOLJkb8

Jeredian, 2020-09-30:
Привет. 
И да и нет. То что плагины работают внутри MainThread Куры - является источником фризов куры, при любом плагине который будет использовать операции занимающие время, либо операции поиска устройств (пингуя их с определенной частотой).
По логике работы плагина принтер должен автоподключаться когда он в сети, поэтому и используется цикл с периодической проверкой доступности принтера.
Есть определенный механизм для авто обнаружения принтеров, который к примеру используется в OctoPrint, но в текущей версии плагина он еще не добавлен. И учитывая что ни MakerBase, ни FlyingBear не имеют заинтересованности в развитии плагина - добавлен хороший функционал может быть не скоро.
Из того можно сделать это можно изменить время с которым плагин проверяет обновления от принтера. Но сейчас это обновление раз в 2 секунды. 
это описано в MKSOutputDevice :

https://github.com/Jeredian/mks-wifi-plugin/blob/6edac06388a3fadbebf769b08ad3cd5991c75d3a/MKSWifiPlugin/MKSOutputDevice.py#L134

 134-137 инициализация таймера

https://github.com/Jeredian/mks-wifi-plugin/blob/6edac06388a3fadbebf769b08ad3cd5991c75d3a/MKSWifiPlugin/MKSOutputDevice.py#L1068

1068-1070 строки это реконнект к принтеру

https://github.com/Jeredian/mks-wifi-plugin/blob/6edac06388a3fadbebf769b08ad3cd5991c75d3a/MKSWifiPlugin/MKSOutputDevice.py#L174

174 строка - ожидание сокета на соединение с принтером.

Собственно блокировка MainThread происходит вызовом self._socket.waitForConnected(2000) . 

Можно попробовать вынести каждый инстанс принтера как отдельный поток и посмотреть как это будет работать.
https://github.com/Jeredian/mks-wifi-plugin/blob/6edac06388a3fadbebf769b08ad3cd5991c75d3a/MKSWifiPlugin/MKSOutputDevicePlugin.py#L152
В файле MKSOutputDevicePlugin.py в строке 152 - находится метод по добавлению принтера и запуску коннекта /поиска текущего принтера в сети.
https://github.com/Jeredian/mks-wifi-plugin/blob/6edac06388a3fadbebf769b08ad3cd5991c75d3a/MKSWifiPlugin/MKSOutputDevice.py#L156
И вот собственно сам метода коннекта к принтеру - с созданием сокета.

closed 2021-01-18 by Jeredian