PrintMakerLab/mks-wifi-plugin #39: Вопрос по имени файла (file name question)

Malderin, 2020-11-13:
briefly: you can make it so that when transferring files via Wi-Fi, the file name is  name.gcode instead of cfffp_name.gcode?
Можно сделать так, чтобы при передаче файлов через вайфай, плагин убирал из имени файла "CFFFP_".
Т.е. имя файла в куре сейчас имеет такой вид: CFFFP_test_name.gcode, хотелось бы чтобы при передаче плагин первым делом менял имя на test_name.gcode ну и далее делал все стандартно.
Объясню зачем это. Недавно мкс на робин нано в.2 наконец сделали передачу файлов по вайфай. Имя в из прошивке ограничено 11 символами. Если я передаю файл с котором именем, т.е. плагин не предлагает мне переименовать файл при старте передачи, то имя файла всегда выглядит одинаково: первые 5 символов названия текущего файла + "-1.gco". Иными словами, для коротких имен файлов имя переданного файла всегда выглядит одинаково CFFFP-1.gco. Т.е. короткие файлы всегда имеют одинаковое имя и на карте памяти принтера не накапливаются а заменяются, что неправильно и неудобно.
Пробовал менять длину имени файла в их прошивке - не удалось, сразу принтер выдает ошибку передачи, почему я не понял, возможно такая особенность есп, не знаю.

Jeredian, 2020-11-13:
cfffp_name.gcode - это настройка самой Cura. Если зайти в меню настроек ,то
можно найти там чекбокс который включает или выключает эту аббревиатуру в
имени файла.

Malderin, 2020-11-13:
Действительно, что то я в эту сторону вообще не смотрел. Спасибо!
--
Отправлено из Mail.ru для Android пятница,

closed 2020-11-13 by Malderin