PrintMakerLab/mks-wifi-plugin #42: Save as TFT file with .3mf defaut extension - dangerous!

GenesioRodrigues, 2020-11-25:
In CURA 4.8.0, when choosing the "save as TFT file" function, the extension shown is .3mf, which creates the risk of overwriting your Cura project. You should have the .gcode as defaut extension.

Elkin-Vasily, 2021-01-17:
Test this issue with the latest plugin version. I cannot repeat this.

GenesioRodrigues, 2021-01-18:
Thanks Elkin, The problem can be replicated if you open a .3mf file, slice
it and choose "Save as TFT file". I checked your code in the
SaveOutputDevice.py file, line 58 (requestWrite function), and there you
coded: last_used_type = self._preferences.getValue ("local_file /
last_used_type"). I think you should set the type "gcode" because at this
point you have already chosen the type "TFT file". All other extensions
should not be shown, because even if another one is chosen, the content of
the file will be of the type .gcode, despite the extension being saved as
chosen. Note that the generated files are the same size, despite different
extensions. Thanks. Genesio

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Elkin-Vasily, 2021-01-18:
I still cannot replicate this issue. With Cura 4.8.0, and the latest plugin version from this repo it looks like:

[image]

What plugin version do you use?

Elkin-Vasily, 2021-01-18:
File extension sets in line 80 of SaveOutputDevice.py

GenesioRodrigues, 2021-01-18:
Sorry, I forgot to inform my OS - Windows 10 PRO x64 Version 2004 Build 19041.746. It looks like the default extension in the combobox is being set to the last extension used by the application. I think at this point the extension should be only .gcode.

Elkin-Vasily, 2021-01-18:
What plugin version do you use?

Still have not luck to replicate this issue. This is on Windows 10 PRO x64:

[image]

GenesioRodrigues, 2021-01-19:
I have tested wiht 1.1.2 (installed from marketplace) and 1.1.3 (downloaded/installed manually from git). Both, if you save your project (.3mf) before, next "Save as TFT file" will show .3mf selected as default extension,  and if you save without change to .gcode extension, the project will be lost, because de content of file will be in gcode format.

Jeredian, 2021-01-19:
Hi, @GenesioRodrigues 
We reproduced this issue
Here there are steps to reproduce it:
1. Open Cura with installed plugin
2. Open STL file
3. Slice model
4. Save project with .3mf file extension 
5. Try to save the STL file by "save as TFT file".

[image]

We will check how it happens and will try to fix it in the current month.

GenesioRodrigues, 2021-01-19:
Ok, thanks.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2021-01-20 by Jeredian

GenesioRodrigues, 2021-01-21:
> Closed #42 <https://github.com/Jeredian/mks-wifi-plugin/issues/42> via #64

Awesome, thanks!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

PR #64 (2021-01-20): Save as TFT file extension fix
resolves #42