PrintMakerLab/mks-wifi-plugin #43: Проблема  с Cura 4.8

Malderin, 2020-12-04:
Board: MKS Robin nano v.2
Sending small files via cura 4.8 does not work

Плата мкс робин нано в.2
На компе установлены 2 куры, 4.8 и 4.7.1
На 4.7.1 все работает нормально. 
Перешел на 4.8 и возникла проблема.
Если посылаешь мелкий файл (например кубик 20х20х20) принтер сразу выпадает в ошибку, пишет ошибка передачи, при чем ошибка появляется моментально. На экране куры при этом не ноль, вернее сразу при нажатии кнопки отправки на печать шкала появляется со значением выше нуля, 3-6% в основном. Или появляется с нулем и моментально перескакивает на 3-6%, может и так.
Заметил так же что если посылать большой файл (5-10 Мб например), перед отправкой появляется полоса загрузки с бегающим ползунком (типа подготовка к отправке), то после этого отправка стартует нормально и все уходит в принтер как положено.
Повторюсь, на 4.7.1 все работает нормально с любыми файлами, проблема с работой плагина именно на 4.8.

Jeredian, 2020-12-04:
Спасибо за баг-репорт. Я постараюсь посмотреть в выходные.
На данный момент я не проверял совместимость с Курой выше 4.7.1. собственно в маркетплейсе будет сегодня релиз для api до этой версии. 
Ты можешь попробовать в plugin.json добавить в supported_sdk_versions - "7.4.0". Возможно это исправит часть возникающих проблем. ( api 7.4.0. - это Cura 4.8.0+ )

Malderin, 2020-12-04:
Конечно попробую, но чуть позже. Если проблема решится - отпишусь.

Malderin, 2020-12-04:
@Jeredian

> Ты можешь попробовать в plugin.json добавить в supported_sdk_versions - "7.4.0". Возможно это исправит часть возникающих проблем. ( api 7.4.0. - это Cura 4.8.0+ )

да, это решило проблему, спасибо!

closed 2020-12-04 by Malderin