PrintMakerLab/mks-wifi-plugin #45: Wrong Preview

LucaCh92, 2020-12-10:
Hi, 
today i'm update the cura wifiplugin to the new 1.1.2 but now the image preview don't works(see img), anyone give me a  plugin previus version?
Thanks
[image]

Jeredian, 2020-12-10:
Yeah, this bug appears after the last makerbase update. If I will have time, I'll solve it at this weekend.
Here is the old version of the plugin.
[MKSWifiPlugin.zip](https://github.com/Jeredian/mks-wifi-plugin/files/5674332/MKSWifiPlugin.zip)
You can found the plugin at  folder C:\Users\{YourUser}\AppData\Roaming\cura\4.6\plugins\MKSWifiPlugin\MKSWifiPlugin . Of course, instead of  "4.6" there will be your version of Cura.
Or you can read this issue: https://github.com/Jeredian/mks-wifi-plugin/issues/11 and make local changes.

LucaCh92, 2020-12-10:
Thank you so much!

misterixxxx, 2020-12-12:
С новой версией еще и сильные тормоза даже в меню, а когда стол с деталью вертишь, время  от времени залипает на несколько секунд. Помогает избавиться от этого полное удаление данного плагина MKS WiFi.

Jeredian, 2020-12-12:
@misterixxxx , да, это тоже известный баг. из-за то того что плагин (спасибо китайцам) изначально в том же потоке что и Кура работает. Как будет возможность обязательно исправлю, ну либо кто нибудь из комьюнити может это сделает, все таки это опен-сорс :-)

Translated version:
yeah, it's a known issue, it's happening because the plugin (thanks for makerbase) using the same thread as Cura, which is very wrong. I will solve it when I have free time, or maybe someones from the community will do that, cause you know, it's open-source.

misterixxxx, 2020-12-12:
Хорошо бы узнать, когда обновленная версия без этого бага появится :)

Jeredian, 2020-12-12:
@misterixxxx , КТТС (как только так сразу).
FlyingBear и Makerbase - это плагин не разрабатывают почти, ну и поддерживать разработчиков тоже не хотят. Так что в свободное время иногда улучшаем.. 
Думаю если завтра хватит времени бета версию сделаю, если не хватит то после нового года.

Translated version:
Well, someday for sure...
FlyingBear and MakerBase - developing this plugin very weak, and they are not interested in supporting developers. For this reason, we fix and upgrade it sometimes in our free time.
I think some part of the issues I will try to fix at the weekend, and others after Christmas or New Year.

dlisin, 2021-01-04:
Looks like problem is here:
https://github.com/Jeredian/mks-wifi-plugin/blob/9efec8fb08e0686433b41c6da9aede40f7c9d1b1/MKSWifiPlugin/MKSOutputDevice.py#L922-L923

For Two Trees Sapphire - `simage 100 / gimage 200` working fine, but for FlyingBear Ghost - correct values are: `simage 50 / gimage 200`

o0-Wolf-0o, 2021-01-12:
> Yeah, this bug appears after the last makerbase update. If I will have time, I'll solve it at this weekend.
> Here is the old version of the plugin. [...]

Hmmmm, right now is 12 January 2021 year SOOOOOOOOOOOO when you will FIX IT !!!!

Jeredian, 2021-01-12:
@o0-Wolf-0o , very soon :-) 
On the first plane, we have an issue with the main thread usage. After it, I will be working on the preview issue.
It's open-source, so ASAP :-)

pash7ka, 2021-01-19:
I confirm, that @dlisin fix works good. I've changed `%APPDATA%\cura\4.8\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevice.py` line 922 to
```
                single_string_file_data += utils.add_screenshot(self._screenShot, 50, 50, ";simage:")
```
and that fixed this issue.

Jeredian, 2021-01-19:
@pash7ka , it will fix the issue only for your type of printer,  for other companies and their printers there still be an issue, because each type of printer has unique firmware and for some reason, we should select or hardcode these params for different printer types.
I described temporary solvation with hardcoding value in #11. It's a very old issue, and I hope that we will fix it in this milestone for January.

closed 2021-02-07 by Jeredian

khaoskosmos, 2021-02-15:
Version 1.2.0 has a garbled preview image, prior to this update the preview worked. Two versions ago I had the split preview as shown in the original post.

[image]

Jeredian, 2021-02-16:
you can try the latest from develop branch and select your side at the plugin settings window.
in 1.2.1 - will select box with base settings and custom settings for some printers.

PR #106 (2021-02-07): MWP-11 Model image fix
closes #11
closes #45
closes #80