PrintMakerLab/mks-wifi-plugin #54: Error: argument 1 has unexpected type 'NoneType'

mikroled, 2021-01-16:
Due to non working transfers with mks robin wifi firmware 1.0.2 I hopefully flashed the latest firmware 1.0.4. to the device.

Now the transfer is startig in Cura 4.8 but the Error "argument 1 has unexpected type 'NoneType' comes up 
during the transfer.
The transfer is nevertheless closed and the printer shows the preview of the model, but the Error message "Err2" is shown on the printer TFT and the print job does't start.

Printer: TwoTrees Sapphire Pro MKS Robin Nano v1.2 with original firmware v2.0

mikroled, 2021-01-17:
I've got it working now by installing the MKS  Plugin from here:
https://github.com/makerbase-mks/Software/tree/master/MKS%20Plugin

Now the Plugin View in Cura shows the **MKS Plugin Connection** twice (verson 1.0.0 grayed out and 1.0.2 active) as shown in the screenshot.
The MKS Wifi Plug is not listet in the Plugin list, but working... 
[image]

Scheinbar habe da wohl was durcheinander gemischt, aber es funktioniert soweit.
Hat jemand eine Idee was passiert ist und wie ich es bereinigen kann?

Jeredian, 2021-01-17:
This thing happens when you have installed the previous version of the plugin manually, or into another folder.
Cura has 2 folders for plugins, one for the plugin from the market place, and another inside the default installation path.

Jeredian, 2021-01-17:
You can remove both of them, and try the latest from this repository.
The old version that you used doesn't have support with Cura 4+.
The latest version from this repo supports Cura 4.6. But you can try it out with Cura 4.8 too.
Also, it's an open-source plugin, feel free to make pull requests to make it better.

mikroled, 2021-01-18:
Ok guys, I removed all plugins and install only the latest from here. Everting works fine now.

Thank you very much for your help and your work on this plugin.

closed 2021-01-18 by mikroled

Elkin-Vasily, 2021-01-18:
This version of plugin will be in Cura market in a few weeks