PrintMakerLab/mks-wifi-plugin #79: [BUG] Can't see the 2nd nozzle

cprobst87, 2021-01-25:
Hello,

I have a MKS GEN_L V1.0 Board with TFT32 and two nozzels. Printing is good with both nozzles, no problems here.

But: I can't see the 2nd nozzle in the cura plugin:
[image]

TFT32, TFTWifi, Config:
[image]

Marlin 2.0.7.2
TFT 3.0.6
Wifi 1.0.4
MKS Cura Plugin V4.4

I can see both extruders on the TFT, only the plugin doesn't show the 2nd. 

Putty connection (to Wifi module at 192.168.0.23) shows RAW:
T:45 /0 B:24 /0 T0:45 /0 T1:23 /0 @:0 B@:0

I think the plugin looks at the first "T" and "B" and not at "T0" and "T1".

Thanks for helping!
Chris

Elkin-Vasily, 2021-01-25:
Try this with the latest commit in develop from this repo. I think it should fix this.

cprobst87, 2021-01-25:
Thank you :-) , but.... it's the same:
[image]

Elkin-Vasily, 2021-01-25:
Ok I will try to figure out what happens.

cprobst87, 2021-01-25:
Ok, I have a solution:
MKSOutputDevice.py, I added this lines (but that's not good for single nozzle machines):
[image]

Also add at line 184 "2" instead of "1": 
 self._number_of_extruders = 2
The same at line 107:
self._num_extruders = 2

Elkin-Vasily, 2021-01-25:
It will be better to use `CuraApplication.getInstance().getGlobalContainerStack().getProperty("machine_extruder_count", "value")` instead of 2

Elkin-Vasily, 2021-01-25:
[image]
And this for section on your screenshot.

I will spend some more time to fix this issue, do not worry)

cprobst87, 2021-01-25:
Great, thank you!

Elkin-Vasily, 2021-01-30:
I think I have fixed this issue, but I have no double extruder printer, so I cannot test it properly. Can you test it for me?

Here is the branch with fixes: https://github.com/Jeredian/mks-wifi-plugin/tree/issue-79

I need you to test:
- e0 and e1 preheat/cancel preheat;
- e0 and e1 jog position.

Elkin-Vasily, 2021-01-30:
I had updated this branch ones more

cprobst87, 2021-01-31:
Prehead is OK, jog works.
Cancel preheat E0 works. E1 cancel but the temperature isn't 0:
[image]

One little thing:
[image]
Because of this mouse-over-effect I have to click several times to get the button...

Elkin-Vasily, 2021-01-31:
Tnx, I know about tooltip bug, but cannot do anything about it right now (it is Cura tooltips, by the way please try this with your printer connected directly with USB cable, does tooltips look the same?).

Reset of the E1 target temperature works same for me, looks like this is a problem in MKS WiFi module... Need to create issue for them... Because we got temperature from here T:45 /0 B:24 /0 T0:45 /0 T1:23 /**202** @:0 B@:0 and it stays 202 for your case.

closed 2021-01-31 by Jeredian