PrintMakerLab/mks-wifi-plugin #81: [BUG] TFT 32 Upload --> Checksum missmatch --> start to early

cprobst87, 2021-01-25:
Hi,
this lines doesn't work with TFT32, I comment out the "self._printFile()". Than upload is OK and manually starting as well, I think it's too early to start the job.
Error from TFT: Checksum missmatch at line: -1

Line 1268
MKSOutputDevice.py

  def _onRequestFinished(self, reply):
        http_status_code = reply.attribute(QNetworkRequest.HttpStatusCodeAttribute)
        self._isSending = True
        self._update_timer.start()
        self._sendCommand("M20")
        preferences = Application.getInstance().getPreferences()
        preferences.addPreference("mkswifi/autoprint", "True")
        if preferences.getValue("mkswifi/autoprint"):
       ##comment out because of checksum error on TFT     
       ## self._printFile()
            return
        if not http_status_code:
            return

Chris

Elkin-Vasily, 2021-10-24:
Is this problem actual with current plugin version?

Elkin-Vasily, 2021-11-21:
closed due to inacktivity

closed 2021-11-21 by Elkin-Vasily