PrintMakerLab/mks-wifi-plugin #113: Can MKS WiFi show and drill down through the directory tree on the SD card?

bellzw, 2021-02-09:
**Is your feature request related to a problem? Please describe.**
When a project has multiple gcode files, it is preferable to put the related files in a single folder in the top level directory.  MKS WiFi presently shows only the gcode files in the top level directory.

**Describe the solution you'd like**
Show files and identify folders.  Clicking a file enables currently shown functions.  Clicking on a folder enables an open (or other appropriate name) button, clicking on which opens the folder and displays files and folders.  Double clicking on a folder should do the same thing as clicking the open button.  Effectively, I'm asking for a file selection dialog box.

**Describe alternatives you've considered**
None.

**Additional context**
I'm using Cura 4.8.0 on a Windows 10 (64 bit) Lenovo Yoga 14 laptop.  My printer is a BIBO2 Dual.  The attached photo has, I think, the information about the TFT display
[image]

Jeredian, 2022-08-31:
@bellzw , do you have this issue on latest version of the plugin?

bellzw, 2022-08-31:
Yes.  I have 1.4.1 installed.  

This what I see after clicking More and then SD Files:
[image]

This is what I see when I select the Print option from the touch screen
[image]
There are no .gcode files in the root, only folders containing .gcode files, 3 .CUR system files (CONFIG, MKS_WIFI, AND MKSTFT28), bak_font and bac_pic system files.  

This is the actual directory of the root of the SD card:
[SD_dir.txt](https://github.com/Jeredian/mks-wifi-plugin/files/9464814/SD_dir.txt)

Jeredian, 2022-09-01:
@bellzw , ok then. I think we need to prioritise this task