PrintMakerLab/mks-wifi-plugin #115: [BUG]1.2.0 MKS WiFi Delete file fails

bellzw, 2021-02-09:
**Describe the bug**
When I display files on the printer's SD card, select one, and click Delete, the file is not deleted.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Cura
2. Connect to printer
3. Click Monitor
4. Click SD Files
5. Select a file (file is highlighted in blue)
6. Click Delete (Open file(s) window disappears)
7. Previously selected file is still present

**Expected behavior**
The selected file should be removed from the SD card.  It would also have been nice if the Open file(s) window did not disappear and simply refreshed.

**Screenshots**
I don't know that screen shots will show anything useful.

**Desktop (please complete the following information):**
 - OS: 64 bit Windows 10 Home Edition running on a Lenovo Yoga 14 laptop
 - Cura version: 4.8.0
 - Plugin version: 1.2.0
 - 3D printer: BIBO2 Dual
 - Motherboard and/or WiFi module firmware version: See photo below
 
[image]

**Additional context**
None.

Elkin-Vasily, 2021-04-25:
Is this issue actual with the latest plugin version?

bellzw, 2021-04-25:
It was with 1.2.0 installed in Cura 4.8.0.  I have since updated Cura to 4.9.0 and MKS to 1.2.2 and see a different problem with file support.  Below is what Cura 4.9.0 shows me when I click SD Files after connecting to the printer.  I have selected file Copy.gcode.
[image]

If I do not select a file, the Delete button does nothing, which is reasonable, although a warning to the effect that the user didn't select a file to delete would be appropriate.

After selecting Copy.gcode,  I click Delete, and then SD Files, I am shows this:
[image]

Note that TWO files have disappeared.  However, if I wait 5 - 15 seconds and click SD Files again, I see
[image]

The Open file(s) dialog box never seems to show all the files if I leave the dialog open and just keep clicking Refresh.  I seem to have to close the dialog box, and then reopen it with SD Files before I will see all 3 files.

After all this, I took the SD card out of the printer and looked at what is in the root folder:
[image]
I have highlighted the 3 files shown to me by SD Files.

I have attached a Cura log generated only from starting Cura 4.9.0, connecting to my printer, clicking SD Files a few times, selecting and trying to delete Copy.gcode, and then clicking SD Files a few times until the Open File(s) window displayed the 3 gcode files in the root directory.  Perhaps it will help you.
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6373381/cura.log)

I noticed a number of M105 M997 commands sent to the printer.  Isn't M997 a command for updating firmware?  Why would you need to do that every 2 seconds?

At line 432, you will see an M30 command with the file path/name being 1:/Copy.gcode, followed by multiple M20 commands.  The returned file lists still have Copy.gcode in them.  Assuming 1:/ is the correct device, does this indicate that my printer's firmware does not implement file manipulation?  I'll be happy to check with the manufacturer (BIBO).

Bottom line: files are still not actually deleted and the plug-in does not display the complete contents of the SD card's root folder, even though it gets the complete list of files in response to an M20.