PrintMakerLab/mks-wifi-plugin #117: File list is empty[BUG]

WEREWOLF1973, 2021-02-10:
**Describe the bug**
On SD Files command files list is not displayed
[image]

**To Reproduce**
Cura 4.8
MKS WIFI Firmware CS 1.0.3
Display MKS TFT35 1.0.7

Elkin-Vasily, 2021-02-10:
What is the plugin version?

WEREWOLF1973, 2021-02-11:
Develop 1.2.1
[image]

Elkin-Vasily, 2021-02-11:
Could you please attach curs log? You can find it here:
- Windows: %APPDATA%\cura\<Cura version>\cura.log , or C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
- OSX: $USER/Library/Application Support/cura/<Cura version>/cura.log
- Ubuntu /Linux: $USER/.local/share/cura/<Cura version>/cura.log

WEREWOLF1973, 2021-02-11:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/5966233/cura.log)

Elkin-Vasily, 2021-02-11:
Looks lite your WiFi module really says that SD contains no files:
```
2021-02-11 17:47:10,079 - DEBUG - [MainThread] mks-wifi-plugin-develop.MKSOutputDevice.write_socket_data [930]: _send_data: M20
2021-02-11 17:47:10,103 - DEBUG - [MainThread] mks-wifi-plugin-develop.MKSOutputDevice.on_read [988]: mks recv: Begin file list
2021-02-11 17:47:10,150 - DEBUG - [MainThread] mks-wifi-plugin-develop.MKSOutputDevice.on_read [988]: mks recv: End file list
2021-02-11 17:47:10,151 - DEBUG - [MainThread] mks-wifi-plugin-develop.MKSOutputDevice.on_read [988]: mks recv: ok
```

Compare this to my log part:
```
2021-02-10 23:48:24,391 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.write_socket_data [886]: _send_data: M20
2021-02-10 23:48:24,439 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: Begin file list
2021-02-10 23:48:24,448 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: .Spotlight-V100.DIR
2021-02-10 23:48:24,460 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Anet-A8-SD2.gcode
2021-02-10 23:48:24,465 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_3DBenchy.gcode
2021-02-10 23:48:24,475 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_3DBenchy1.gcode
2021-02-10 23:48:24,486 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Door Lock Temp v3.gcode
2021-02-10 23:48:24,491 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_pulley75.gcode
2021-02-10 23:48:24,501 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_skrabka.gcode
2021-02-10 23:48:24,508 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_helmet.gcode
2021-02-10 23:48:24,519 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_armour.gcodeplain=
2021-02-10 23:48:24,532 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_armour.gcode
2021-02-10 23:48:24,536 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_armour1.gcode
2021-02-10 23:48:24,547 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center.gcode
2021-02-10 23:48:24,560 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center_1.gcode
2021-02-10 23:48:24,574 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center_2.gcode
2021-02-10 23:48:24,587 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center_3.gcode
2021-02-10 23:48:24,601 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center_4.gcode
2021-02-10 23:48:24,614 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: FBG5_Center_5.gcode
2021-02-10 23:48:24,627 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: End file list
2021-02-10 23:48:24,634 - DEBUG - [MainThread] MKSWiFiPlugin.MKSOutputDevice.on_read [944]: mks recv: ok

```

Pay attention that no dirs will be displayed, only root files. Does printer itself find any files in SD root?

Do you have the latest WiFi module firmware?

WEREWOLF1973, 2021-02-11:
Does printer itself find any files in SD root?
**Yes. Moreover, files are sended to printer from cura directly to print correctly. Afterwards files are holded on SD and visible through TFT menu browser (with TFT preview picture)**
Do you have the latest WiFi module firmware?
**CS 1.0.3 - Last known**

Elkin-Vasily, 2021-02-11:
What SD card size do you have?

WEREWOLF1973, 2021-02-11:
Problem has been solved after TFT35 firmware upgrade to 1.0.8
[image]
[image]
Thank to attention!!

WEREWOLF1973, 2021-02-11:
MKS TFT35 firmware should be upgraded to 1.0.8

closed 2021-02-11 by WEREWOLF1973