PrintMakerLab/mks-wifi-plugin #122: [FEATURE] OctoPrint Support

jneilliii, 2021-02-13:
**Is your feature request related to a problem? Please describe.**
Not really a problem, but I've written a couple thumbnail plugins for Cura/PrusaSlicer/SuperSlicer and came across this new embedded thumbnail format and would like to extend my Cura Thumbnails plugin to support it. 

**Describe the solution you'd like**
A little assistance on the image embed format. I understand that you are creating a png snapshot [here](https://github.com/Jeredian/mks-wifi-plugin/blob/5dd1ea3da2fe1a186092bcd2ba34ce6f96f6e673/utils.py#L43-L69) and wanted to verify that this function is converting rgb pixel values into html hex strings?

**Describe alternatives you've considered**
Not sure what other format it could be.

**Additional context**
Would it be possible to get some additional context or possibly a hand on a conversion function that would give me the reverse of this function (c array to png I think).

Elkin-Vasily, 2021-02-14:
Not really understood what do you want to do.

Do you want to take a part of code that makes images for MKS TFTs to your plugin? And you need to understand what functions do you need to do so?

jneilliii, 2021-02-14:
I'm trying to reverse the conversion process. The way I understand your plugin to work now is that it takes a snapshot (assuming in png format as that is what the ufp format file does) and then converts it into TFT format to embed inside the gcode. I want to take the lines added in the gcode and convert it back to png file to store in OctoPrint and provide those thumbnails in the web UI. It might make more sense if you look at my Cura Thumbnails plugin repo [here](https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage). I can allow OctoPrint to upload tft format files and process them on upload to extract the thumbnail.

Jeredian, 2021-02-14:
Hi, @jneilliii 
If I understand you right you need to make full reverse decoding from gcode hex-included line to RGB tuple and to a png file.
I think it's mostly related to the method that saves png to the hex. We still use not so good design with bitwise operators, and I'm not sure is it a true hex value or its byte representation.
But to clarify all operation we need to realize these 3 steps:
1. Read line by line and separate by some Regex each hex value
2. Convert hex to the RGB tuple, f.ex. if you will have a simple hex, you can use something from this StackOverflow question or try to use colormap
https://stackoverflow.com/questions/29643352/converting-hex-to-rgb-value-in-python
3.Convert an array of all tuples to a png file, for this step I found another question at the StackOverflow, which may help you a little bit.
https://stackoverflow.com/questions/47962048/python-creating-an-image-from-rgb-tuples

Jeredian, 2021-02-14:
Also, we use conversion from RGB888 to RGB565. 
```
r = pixel_color.red() >> 3
g = pixel_color.green() >> 2
b = pixel_color.blue() >> 3
```
So if you have hex with RGB585 you will need to make some bitwise operations too. Something like that:
```
//RGB565 -> RGB888 using integer approximation
r8 = ( r5 << 3 ) | (r5 >> 2);
g8 = ( g6 << 2 ) | (g6 >> 4);
b8 = ( b5 << 3 ) | (b5 >> 2);
```

jneilliii, 2021-02-14:
Awesome, your input and links are a huge help. Once I've had some time to digest I may post back in here if I run into any weird issues or need further guidance. The RGB format conversion was definitely something I wouldn't have figured out right away.

closed 2021-02-14 by jneilliii