PrintMakerLab/mks-wifi-plugin #127: [FEATURE]

willpuckett, 2021-02-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

This might be a long one, but I thought it would be worth mentioning. There's a obviously a lot of domain knowledge in this repository about MKS. 

Currently, I install Octoprint to manage and monitor my printers in Home Assistant. This lets me log be easily monitor several printers at once, and do things like turn them off when they finish printing.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

It would be exhilarating to see this project ported/refactored into a home assistant integration that would expose sensor data to home assistant. Including gcode upload and job progress/ control would be icing on the cake.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I currently use octoprint to deal with this, but reducing cost and using the already on board hardware would be preferable be more environmentally friendly. 

**Additional context**
Add any other context or screenshots about the feature request here.

This is the current octoprint equivalent to what it would be exciting to see with MKS: https://github.com/cmroche/OctoPrint-HomeAssistant

Jeredian, 2021-02-17:
Hi, @willpuckett .
That's a great idea, let's keep it for the future. Maybe someone who already has Home Assistant will start development in that way.

gumileeev, 2021-05-04:
Здравствуйте! Тоже очень хотел бы видеть интеграцию в homeassistant для flying bear ghost 5. Если homeassistant будет получать статусы от приртера, будет легко сделать польщование приртером комфортнее: получать уведомления в telegram об окончании работы, уведомления об ошибках, отключать вытяжку и другое.

Elkin-Vasily, 2021-10-31:
There is such project by @Blooof.

[Here it is.](https://github.com/Blooof/Mks2MqttReporter)

closed 2021-10-31 by Elkin-Vasily