PrintMakerLab/mks-wifi-plugin #128: [BUG]Исчезает окно передачи файла в куре

Malderin, 2021-02-25:
Не понимаю в чем дело, но в куре 4.8 при отправке на печать файла, окно передачи (то что с галкой "начать печать автоматически после передачи") появляется бувально на секунду и исчезает. Галку при таком раскладе снять нереально. При этом на экране принтера появляется стандартный интерфейс передачи файла, файл передается и печать стартует (галка видимо стоит).
Есть мысли что сделать? Плагин переустанавливал.

Malderin, 2021-02-25:
Забыл. Для чистоты эксперимента поставил куру с нуля, добавил профиль эндер-2, установил из маркета вайфай плагин. Никакие настройки не трогал - проблема осталась.
Возможно это важно, раньше на вин7 проблем не было, сейчас на вин 10 перешел и эта проблема появилась.

Jeredian, 2021-02-26:
@Malderin ,
Для того чтобы было возможно понять Вашу проблему необходимы дополнительные данные:
**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 11.1]
 - Cura version: [e.g. 4.8.0]
 - Plugin version: [e.g. 1.1.2]
 - 3D printer: [e.g. FlyingBear Ghost 5]
 - Motherboard and/or WiFi module firmware version: [e.g. V5.8/V5.8]
 
 А также необходим **файл логов**:
 - Windows: %APPDATA%\cura\<Cura version>\cura.log , or C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
 - OSX: $USER/Library/Application Support/cura/<Cura version>/cura.log
 - Ubuntu / Linux:$USER/.local/share/cura/<Cura version>/cura.log

> Eng
> In order to be able to understand your problem, **additional data** and **log file** are needed.

Malderin, 2021-02-27:
> Для того чтобы было возможно понять Вашу проблему необходимы дополнительные данные:
> **Desktop (please complete the following information):**

**OS:** 
Выпуск	Windows 10 Pro
Версия	20H2
Дата установки	‎08.‎02.‎2021
Сборка ОС	19042.685
Взаимодействие	Windows Feature Experience Pack 120.2212.551.0
**Cura version:** 4.8.0
**Plugin version:** 1.2.0
**3D printer**: KingRoon KP3 (Хотя это значения не имеет, т.к. плата там не родная)
**Motherboard and/or WiFi module firmware version:** : MKS Robin nano v.2, а вот прошивка вайфай модуля стояла с завода, ничего не менял и какая версия даже и не знаю. При покупке на сайте продавца было сказано что прошивка залита и обновлять ничего не нужно.

> А также необходим **файл логов**:

[cura.zip](https://github.com/Jeredian/mks-wifi-plugin/files/6053836/cura.zip)

И вот как это выглядит, видос снял:
https://yadi.sk/i/mqMNN3_ORL19qA

Malderin, 2021-02-27:
Обновил модуль вайфай до последней версией прошивки с гитхаба (1.0.4) - никаких изменений

Jeredian, 2021-02-28:
@Malderin , Судя по видео, у Вас просто очень быстро передаются файлы. т.к. сразу после передачи началась печать.
Попробуйте для сравнения большую и сложную модель перекинуть, тогда у вас будет сообщение об отправке не сразу исчезать. Дайте знать о том какое поведение будет в данном случае.

На данный момент есть два варинта решения задачи:
1. Добавить задержку перед отправкой на печать, в виде 1-2 секунд по умолчанию, и настройкой в окне плагина. 
2. Вынести чекбокс с началом печати в настройки плагина чтобы настроить это в статичном окне и для любой отправки на печать.

> Eng
> @Malderin, From your video, it looks like your files are transferring very quickly because right after the transfer, the printing was started. Try to transfer a large and complex model for comparison, then you will have the sending message not immediately disappear. And please, let me know what the behavior will be in this case.
> 
> At the moment there are two possible solutions to the issue:
> 1. Add a delay before sending to print, in 1-2 seconds by default, and implement this setting in the plugin window. 
> 2. Put a checkbox with the beginning of printing in the settings of the plugin to configure it in a static window and for any sending to print.

Malderin, 2021-02-28:
> Судя по видео, у Вас просто очень быстро передаются файлы

Ах если б это было так. На самом деле в начале появляется просто бегающая полоса (типа подготовки к загрузке) - ее и видно на видео, потом должна появиться полоса загрузки, но она не появляется, сразу проскакивает на окно типа все передано. 

> 2\. Вынести чекбокс с началом печати в настройки плагина чтобы настроить это в статичном окне и для любой отправки на печать.
Вот реально огромная модель:
https://yadi.sk/i/SQSi-EOKS51yDA
Экран принтера не снял, но там еще пару минут висело окно передачи, прежде чем появился диалог печати (если что могу переснять).

А вот это было б хорошо вообще в любом случае, даже без привязки к моей проблеме.
П. 1 проблему не решит, она не в этом как я написал выше, но для опции настройки плагина, как по мне, не помешал бы.

Jeredian, 2021-02-28:
Согласен, исходя из второго видео, у нас процесс отправки модели на принтер не соответствует реальной скорости отправки.

Так что да, будет комплексное решение:
1. Проверить процесс отправки, и выяснить почему он завершается до рельного получения файла на принтере, т.к потенциально могут буть проблемы начала печати, при закрытии Cura, либо отправки команды на печать, до того как файл был отправлен.
2. Вынести чекбокс для отправки на печать в настройки. т.к. потенциально пользователь может не успеть принять решение при отправке файла, либо изменит его когда файл уже отправлен.

> Eng
> I agree, based on the second video, our process of sending gcode file to the printer not according to the actual speed of sending.
> 
> So, yeah, better will be making a complex solution:
> 1. Check the sending process and find out why it completes before the file is received on the printer. There can be potential issues with print starting when Cura closed after the file sent, or with sending a command to print before the file was received on the printer.
> 2. Add checkbox to "start print" in the plugin settings. Because there can be the case when a user may not have time to make a decision or may try to change settings when a command to print complete

Malderin, 2021-02-28:
Если нужны какие то дополнительные данные или логи, я всегда готов все предоставить.

Elkin-Vasily, 2021-04-25:
Судя по логу у Вас происходит следующее:

2021-02-27 09:29:30,215 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [840]: Upload _onUploadProgress bytes_sent 16384
2021-02-27 09:29:30,215 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [842]: Upload _onUploadProgress bytes_total 399091
2021-02-27 09:29:30,221 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [840]: Upload _onUploadProgress bytes_sent 399091
2021-02-27 09:29:30,221 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [842]: Upload _onUploadProgress bytes_total 399091
2021-02-27 09:29:30,229 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2021-02-27 09:29:30,235 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2021-02-27 09:29:30,247 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/plugins/PreviewStage/PreviewMenu.qml:25: TypeError: Cannot read property of null
2021-02-27 09:29:30,252 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/plugins/PreviewStage/PreviewMenu.qml:26: TypeError: Cannot read property of null
2021-02-27 09:29:30,380 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2021-02-27 09:29:31,060 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-02-27 09:29:34,549 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [840]: Upload _onUploadProgress bytes_sent 0
2021-02-27 09:29:34,550 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._onUploadProgress [842]: Upload _onUploadProgress bytes_total 0

Два места где происходит передача и она проскакивает мгновенно в одном случае, в другом вообще пытаемся передать 0 байт.

Попробуйте удалить файл лога, запустите Куру и попередавать файлы пару раз и приложите лог.

Malderin, 2021-04-25:
> Попробуйте удалить файл лога, запустите Куру и попередавать файлы пару раз и приложите лог.

Сделал. Плагин версии 1.2.2 (обновился сегодня и маркета)
[cura.zip](https://github.com/Jeredian/mks-wifi-plugin/files/6372383/cura.zip)

Elkin-Vasily, 2021-04-25:
Тааак:
DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1012]: mks recv: M994 /NTITL~1.GCO;896262

А SD-карта у Вас какая? Штатная? Попробуйте с хорошей SD (из тех что не шли с чем-то китайским в комплекте).

Malderin, 2021-04-25:
> А SD-карта у Вас какая? Штатная? Попробуйте с хорошей SD (из тех что не шли с чем-то китайским в комплекте).

Это было первое что я попробовал. Перепробовал все китайские что были, из нормальных были карты трансцендент и самсунг. Не меняется ничего.