PrintMakerLab/mks-wifi-plugin #129: BUG? Cura and printer connection, connection procedure

Malderin, 2021-02-28:
Не соединяется Кура и принтер. 
Порядок такой:
1. Запускаем Куру
2. Включаем принтер 
3. Ждем
4. Соединения не происходит, на вкладке "Монитор" пусто, нет ни кнопок от плагина ни температур, ничего.

Для возможности и упрощения соединения в такой ситуации создал второй принтер (Эндер -2, выбрал из готовых). После выполнения вышеуказанных действий, переключаюсь на эндер 2 и сразу назад на свой принтер, плагин появляется, принтер обнаруживается и соединяется, вот демонстрация:
https://disk.yandex.ru/i/7E_vLEkCJcBJfw

Если изменить порядок и сперва включить принтер, потом запустить куру, то все подхватывается и соединияется.

Elkin-Vasily, 2021-02-28:
Это известный баг, будет решен в рамках работы над #58

Jeredian, 2021-02-28:
duplicate #58

Jeredian, 2021-02-28:
Duplicate of #58

PR #236 (2021-11-27): [MWP-129] Connection procedure when printer start before Cura fix
closes #129
closed 2021-11-27 by Jeredian