PrintMakerLab/mks-wifi-plugin #134: [FEATURE] Android app for 3D print monitoring

Medvediy, 2021-03-02:
**Is your feature request related to a problem? Please describe.**
Sometimes we need to control 3D Printing parameters while Cura and PC are not powered on. 

**Describe the solution you'd like**
So it will be great to got Android App which will connect to printer as MKS wifi plugin via IP and will show all the same parameters. It will help to control printing from the any place you are. 
Appreciate your job and all you're doing for us!

Elkin-Vasily, 2021-04-25:
Is it the app you are searching for?
https://github.com/makerbase-mks/Software/tree/master/MKS%20Control%20-Standalone%20android%20app

Imperattor, 2022-04-13:
> Is it the app you are searching for? https://github.com/makerbase-mks/Software/tree/master/MKS%20Control%20-Standalone%20android%20app

language only  China  !!! ((((

Jeredian, 2023-10-31:
@Medvediy , @Imperattor  are you still interested ?

Basically, we can create a simple application. It will certainly work on a local network, but you will need a white IP and possibly port forwarding to monitor the 3D printer outside of that network.