PrintMakerLab/mks-wifi-plugin #135: [BUG] WIFI data corruption on gcode

xaff, 2021-03-05:
Hello and thanks for your work :)

**Describe the bug**
When I send with Cura something to print via wifi, the upload seems ok but there is network errors in gcode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Prepare'
2. Click on 'Slice then Print over Wanhao D12'
3. Upload ok but print fails
4. Get File on SD card
5. Compare with the one directly saved on windows by cura
6. See transfer errors on gcode

**Screenshots**
Comparison in Text:
[image]
in Hex:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Cura version:  4.8.0
 - Plugin version: 1.2.1
 - 3D printer: Whanao D12
 - Motherboard and/or WiFi module firmware version: Wanhao robin_nano35 V1.061B-D12/230

Here as the gcodes:
[gcodes.zip](https://github.com/Jeredian/mks-wifi-plugin/files/6089428/gcodes.zip)

Maybe add a CRC check on network packet ?

xaff, 2021-03-13:
Hello,

I finally figured it out, it was the SD card, wich is quite strange because it's a new one and it work well on computer but not on the printer !?

Anyway, so it has nothing to do with the wifi.
Very sorry for the inconvenience

Please delete this Issue.

Best regards.

closed 2021-03-13 by xaff

Elkin-Vasily, 2021-03-13:
What size of SD it was?

xaff, 2021-03-14:
4G, the one I got with the printer.