PrintMakerLab/mks-wifi-plugin #138: TFT info (image) is not available

ColdBeer72, 2021-03-12:
Since last update (1.2.1) on march 10th, there is no TFT info (no image to show on TFT35 display) on the generated gcode file.

Elkin-Vasily, 2021-03-12:
Go to "MKS WiFi Connection" settings and set your printer preview size. It turned off by default.

closed 2021-03-12 by Elkin-Vasily

ColdBeer72, 2021-03-12:
> Go to "MKS WiFi Connection" settings and set your printer preview size. It turned off by default.

My printer is not connected through WiFi, just through "human movement" (with the SD card)

Elkin-Vasily, 2021-03-13:
Just add random IP address, then you can set image size.

We have created issue to manage image size without adding IP (#136)

ColdBeer72, 2021-03-13:
Fine! I used 127.0.0.1 (localhost) as IP address but, I suppose **simage** is "icon view" and **gimage** is "printing view". How do I know which is the correct one for Robin Nano V1.2?

Elkin-Vasily, 2021-03-17:
Default wil be ok for you

Boneunit, 2021-03-22:
Hi,
Thank you for your help but it doesn't work for me.
I don't think that it's a problem of printer settings, it comes from the curas output file which doesn't contain any image preview

rdulak, 2021-04-23:
> 
> 
> Fine! I used 127.0.0.1 (localhost) as IP address but, I suppose **simage** is "icon view" and **gimage** is "printing view". How do I know which is the correct one for Robin Nano V1.2?

How did you add the ip address to an existing printer ?

ColdBeer72, 2021-04-23:
> > Fine! I used 127.0.0.1 (localhost) as IP address but, I suppose **simage** is "icon view" and **gimage** is "printing view". How do I know which is the correct one for Robin Nano V1.2?
> 
> How did you add the ip address to an existing printer ?

In printer configuration, I don't have my computer at this moment, but you'll see something like "network config" in the selected printer.