PrintMakerLab/mks-wifi-plugin #141: Uploading error [BUG]

kazakh1979, 2021-03-16:
Before the last update I had no issue with this plugin. But after the update of march 10th I get everytime "upload error" message.

Jeredian, 2021-04-24:
@kazakh1979 , is it still actual for you?
If it is, please get to us more information, you can see it below. 

**Describe the bug**: clear and concise description of what the bug is.

**Reproduce Steps** :to reproduce the behavior:
Go to '...'
Click on '....'
Scroll down to '....'
See error

**Expected behavior**: clear and concise description of what you expected to happen.
**Screenshots**: If applicable, add screenshots to help explain your problem.
**Desktop configuration**:
OS: [e.g. Windows 10, macOS 11.1]
Cura version: [e.g. 4.8.0]
Plugin version: [e.g. 1.1.2]
3D printer: [e.g. FlyingBear Ghost 5]

Elkin-Vasily, 2021-05-28:
Is this issue actual?

Levingard, 2021-06-09:
I have exatly the same problem. I can connect to Printer via wifi, move axis, monitor prints, but when i try to send the printjob via wifi, the upload window disappears and the printer says uploading error.

Printer: Flying Bear Ghost 5 with actual firmware and Cura 4.9.1

Markosyan-AR, 2021-06-17:
The same issue with flying bear ghost 5 + cura 4.9.0 + 1.2.2 plugin. Failed to upload large files.
Printer Firmware- [https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)
Is it possible to collect tracing info or logs or something like that?

idlesign, 2021-09-18:
I was able to reproduce it on Ghost 5 & Cura 4.11.0 when the repo contents is copies to plugins dir manually: sending model with wi-fi leads to Uploading error on display.

Using plugin 1.2.4 from marketplace **works as expected** though.

Elkin-Vasily, 2021-10-24:
> **Updated.**
> 
> I was able to reproduce it on Ghost 5 & Cura 4.11.0, plugin 1.2.4 without SD card: sending model with wi-fi leads to Uploading error on display. With SD card **works as expected** though.
> 
> 

WiFi module saves files on SD and then start printing. Without SD it won't work.

Elkin-Vasily, 2021-11-21:
closed due to inacktivity

closed 2021-11-21 by Elkin-Vasily