PrintMakerLab/mks-wifi-plugin #142: [BUG]

martinbogo, 2021-03-19:
**Describe the bug**

After installing the MKS WiFi plugin in Cura 4.8.0 on Windows 10... I see the "Save as TFT" option, however the file that is produced lacks the simage; image preview... only the gcode is output.

**To Reproduce**
1. Download Cura 4.8.0
2. Clone the mks-wifi-plugin repo from https://github.com/Jeredian/mks-wifi-plugin
3. Slice calibration cube or any other stl file with standard settings
4. Select "Save as TFT" from the dialog to the local disk or SD card
5. Open the resulting .gcode file, and there is no image preview

**Expected behavior**

The beginning of the file should contain an image preview rendered as a series of simage; lines which are rendered by the MKS TFT35 LittleVGL interface in Marlin 2.0. 

**Desktop (please complete the following information):**
 - OS: Windows 10, MacOS Big Sur Beta 4
 - Cura version: 4.8.0
 - Plugin version: both develop branch, and 1.1.2
 - 3D printer: FOKOOS Odin 5-F3
 - MKS Robin Nano v1.2

Boneunit, 2021-03-21:
Hi,
I have exactly the same problem. It works sometime but in most case their is no preview picture.

Elkin-Vasily, 2021-03-22:
Hi, in the latest version of plugin preview image is set to "None" by default.
You need to add some (any) IP to printer, and set "simage" and "gimage" options to yours TFT screen (if you don't now what it is most likely "Default" will be ok for you).

Elkin-Vasily, 2021-03-22:
duplicates #138

Boneunit, 2021-03-22:
Hi,
Thank you for your help but it doesn't work for me.
I don't think that it's a problem of printer settings, it comes from the curas output file which doesn't contain any image preview

ColdBeer72, 2021-03-23:
It worked for me, just adding an IP (127.0.0.1, which is localhost) to one of your printers in Cura. You don't need to add IP to the rest of your real or virtual printers. After that, every "TFT/gcode" file has the image inside.

Elkin-Vasily, 2021-03-23:
> Hi,
> 
> Thank you for your help but it doesn't work for me.
> 
> I don't think that it's a problem of printer settings, it comes from the curas output file which doesn't contain any image preview

Do you have some another plugins installed? For example Arc Welder can cause this issue.

Boneunit, 2021-03-26:
Hi,
I have uninstalled all the plugins and it's still the same thing no image preview in the GCode.

Elkin-Vasily, 2021-03-26:
Could you download here a log file? It can be found:

• Windows: %APPDATA%\cura<Cura version>\cura.log , or C:\Users\\AppData\Roaming\cura<Cura version>\cura.log
• OSX: $USER/Library/Application Support/cura//cura.log
• Linux: $USER/.local/share/cura//cura.log

Boneunit, 2021-03-27:
Here is the log file :
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6215127/cura.log)

Boneunit, 2021-04-01:
Hi,
Any idea ?

Boneunit, 2021-04-24:
Hi,
Thanks for your answer. I have'nt the plugin settings button in my printer settings

Jeredian, 2021-04-24:
@Boneunit , you are right, you need to press on "MKS wifi connection", after that you will see plugin settings window and "image settings" button. Please, take a look at the step-by-step screenshoots:
1. Press to the "MKS wifi connection" button
[image]
2. Press to the image settings" button
[image]
3. Select default image preview setup and save it by clicking "ok" button.
[image]
4. After it you will see that settings saved
[image]
5. Close plugin settings window.
6. Test with some random 3d model...

Hope it will helps to you

Boneunit, 2021-04-24:
Thanks a lot for your help. I have done each step but it's still the same 
[image]

Jeredian, 2021-04-24:
@Boneunit , can you attach gcode file and log with saving it? I'll check it...

Boneunit, 2021-04-25:
Here are the log file and the gcode file:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6371316/cura.log)
[Untitled.gcode.txt](https://github.com/Jeredian/mks-wifi-plugin/files/6371317/Untitled.gcode.txt)

Elkin-Vasily, 2021-04-25:
> Here is the log file :
> [cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6215127/cura.log)

In your log I see this:
DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevicePlugin.reCheckConnections [149]: GlobalContainerStack change None

It is some strange behaviour... Try to delete your printer from Cura and add it once more.

Elkin-Vasily, 2021-04-25:
> Here are the log file and the gcode file:
> [cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6371316/cura.log)
> [Untitled.gcode.txt](https://github.com/Jeredian/mks-wifi-plugin/files/6371317/Untitled.gcode.txt)

In this log you have:
DEBUG - [MainThread] MKSWifiPlugin.__matchVersion [56]: This version of MKS WiFi Plugin is not compatible with current version of Cura. Please check for an update.

Jeredian, 2021-04-25:
@Boneunit , if you are using Cura 4.9.0 at now, I'd suggest to manually update the plugin to 1.2.2 RC1 ( https://github.com/Jeredian/mks-wifi-plugin/releases/tag/1.2.2-dev )

Boneunit, 2021-04-25:
I have done the update with the new version 1.2.2 and it's still the same

Elkin-Vasily, 2021-04-25:
Please attach log file with latest plugin version installed

Ryszard-wpl, 2021-04-25:
I have the same problem. 
In step 3 of your guide - I cannot select any options - all are grey and inactive. Plugin is connected to printer and there is MKS TFT35. 
[image]

Boneunit, 2021-04-26:
Here is the log file 
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6374418/cura.log)

Jeredian, 2021-04-26:
@Ryszard-wpl , it's another issue, I created a ticket for it #163. And we will try to solve it ASAP.

Jeredian, 2021-04-26:
@Boneunit, thanks for sending the new log file, but it seems that you save the file by using the default Cura saving. 
Can you send to us a screenshot of which button do you use to save the gcode file?

Boneunit, 2021-04-26:
This is this button 
[image]

Elkin-Vasily, 2021-05-27:
> This is this button 
> 
> [image]
> 
> 

Do you have any other plugins installed?

Boneunit, 2021-05-28:
Hi,
Here is the plugin list : 
[image]

Boneunit, 2021-08-28:
Hi,
I have installed the 1.2.4 and it's still the same.

Jeredian, 2021-08-28:
Hi, @Boneunit, 
can you attach, please, the latest log file, with a full process ( loading cura, slice file, save it with our plugin)?
And as I saw, it seems we forgot to ask about your printer model...
What happens if you send a file via WIFI  to the printer?

Boneunit, 2021-08-29:
Hi,
Here is the log file :
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7071844/cura.log)
My printer is a Ender 3.
I don't have the wifi on my printer

Jeredian, 2021-09-10:
@Boneunit , 
Hi, I just back from vacation and checked your file.
Maybe there can be some issue with the slicing process and somehow plugin doesn't encapsulate this. (stuff like on 1272 line),  but the chance of it is low.

Have you tried to re-check the "screenshot support" settings? it seems that we have an issue with correct saving these settings, and for some users, it switched back to disabled or default. It should be like this:
[image]

Also, can you try to select custom settings, and re-select the default one? I know it looks ridiculous but if we don't have any error about saving, there should be something wrong with the settings parameter.

Jeredian, 2021-09-10:
@Boneunit, 
I added extra logs for file saving, can you try to use the latest version from the `develop` branch, please, and send a new `cura.log` file?
Btw, you can try Cura 4.11, a plugin from the `develop` branch compatible with it.

Elkin-Vasily, 2021-11-21:
closed due to inacktivity

closed 2021-11-21 by Elkin-Vasily