PrintMakerLab/mks-wifi-plugin #143: [BUG]

Dtcreation, 2021-03-23:
**Describe the bug**
My printer reboot each time the plugin connect to the printer (Cura 4.8)

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. The printer reboot

**Desktop (please complete the following information):**
 - OS: WIndows 10
 - Cura version: 4.8.0
 - Plugin version: 1.2.1
 - 3D printer: Wanhao D12
 - Motherboard and/or WiFi module firmware version: MKS Nano Robin 35 + MKS Wifi

**Additional context**
Add any other context about the problem here.

Elkin-Vasily, 2021-03-23:
Can you specify mainboard and WiFi module firmware?

Dtcreation, 2021-03-23:
MKS Robin Nano V1.2
[image]

Elkin-Vasily, 2021-03-26:
It is a hardware version. And what about firmware?

Elkin-Vasily, 2021-04-25:
Is this issue actual?

closed 2021-05-28 by Elkin-Vasily