PrintMakerLab/mks-wifi-plugin #144: [BUG]

X-Dron, 2021-03-28:
**Describe the bug**
Don't work in Ultimaker Cura Arachne_engine_beta 99.9.1

**To Reproduce**
Install Cura Arachne_engine_beta 99.9.1
Check printer settinngs

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 7,]
 - Cura version: [e.g. Cura Arachne_engine_beta 99.9.1]
 - Plugin version: [e.g. 1.2.1]
 - 3D printer: [e.g. FlyingBear Ghost 4S]
 - Motherboard and/or WiFi module firmware version: [Marlin 2.0.7.2]
[image]

X-Dron, 2021-03-28:
fix
    if cura_version == "Arachne_engine_alpha" or cura_version == "Arachne_engine_beta":

Elkin-Vasily, 2021-03-29:
Hi, just get latest commit from develop it was fixed already.

closed 2021-03-29 by Elkin-Vasily