PrintMakerLab/mks-wifi-plugin #146: get data from 3d Printer

ivale, 2021-04-09:
Hy, I would like to access the monitor data of my FlyingBear Ghost 5 while running.
The plugin works perfectly in Cura but I would like to write a simple python script that works outside of Cura and take the same monitor data. 

Can you help me? :)

Elkin-Vasily, 2021-04-10:
Hmmm, do you want to write it from the scratch or take our plugin as a base? I think the best way is the first. If you agree, here is MLS-WIFI protocol: https://github.com/makerbase-mks/MKS-WIFI/blob/master/Transfer%20protocal%20%20between%20ESP%20and%20MCU.docx

Is it that you need?

ivale, 2021-04-12:
I was wondering if there's someone with an example code from where to start.
And yes, the MKS-WIFI protocol is the right way to make it, thanks.

closed 2021-04-14 by Jeredian