PrintMakerLab/mks-wifi-plugin #148: [BUG] add support for 4.9.0 beta+

darkblaze69, 2021-04-17:
Doesn't work in recently released Cura 4.9.0 beta. Please add support.

khaoskosmos, 2021-04-18:
Same problem... from the log:

`2021-04-18 14:32:18,034 - DEBUG - [MainThread] MKSWifiPlugin.__matchVersion [56]: This version of MKS WiFi Plugin is not compatible with current version of Cura. Please check for an update.
2021-04-18 14:32:18,034 - WARNING - [MainThread] MKSWifiPlugin.register [28]: Plugin not loaded because of a version mismatch
2021-04-18 14:32:18,036 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [445]: Plugin MKSWifiPlugin did not return any objects to register`

khaoskosmos, 2021-04-18:
Okay, the fix is pretty easy.

Go to the cura plugins directory (I'm on Mac, so it was ~/Library/Application\ Support/cura/4.9/plugins/MKSWifiPlugin/

Open "plugin.json" and replace the line that says "maximum_cura_version": "4.8" to 4.9

What it should look like:
[plugin.json.zip](https://github.com/Jeredian/mks-wifi-plugin/files/6331962/plugin.json.zip)

`{
    "name": "MKS Plugin Connection",
    "author": "Makerbase, Jeredian, Elkin-Vasily",
    "version": "1.2.1",
    "minimum_cura_version": "4.5",
    "maximum_cura_version": "4.9",
    "description": "Сonnect to MKS Robin WiFi/MKS TFT WiFi in Cura",
    "api": "7.1.0",
    "supported_sdk_versions": [
        "7.1.0",
        "7.2.0",
        "7.3.0",
        "7.4.0"
    ],
    "i18n-catalog": "mksplugin"
}`

Jeredian, 2021-04-19:
@darkblaze69 , @khaoskosmos , We will update the supporting new Cura version at the next release because there are a lot of API changes and we might need to test how the plugin works with them.
You can use a quick workaround that @khaoskosmos suggested, but use it at your risk, and if there will be issues with the plugin please report them with describing Cura version.

closed 2021-04-24 by Jeredian

jellespijker, 2021-04-25:
The package is approved and ready to be downloaded/upgraded from the marketplace

PR #155 (2021-04-24): MWP-148 adding Cura 4.9 support
closes #148