PrintMakerLab/mks-wifi-plugin #149: [BUG]Cant make Cura connect to my printer with MKS WiFi Plugin installed

aykunsezer, 2021-04-19:
I have an Ender 5 Plus with MKS SGenL v2.0 and MKS TFT35. I have wifi enabled and connected to local wifi. My I am using Cura 4.8 with the plugin installed. However Cura does not seem to find the printer and cannot transfer both the gcode nor tft.

Jeredian, 2021-04-23:
@aykunsezer , please provide a log file, to check is ok or not with the process after you press the "connect" button.
It can be found here:
```
• Windows: %APPDATA%\cura<Cura version>\cura.log , or C:\Users\\AppData\Roaming\cura<Cura version>\cura.log
• OSX: $USER/Library/Application Support/cura//cura.log
• Ubuntu /Linux: $USER/.local/share/cura//cura.log
```
And describe all your steps, please, so we can reproduce and help you better. 
Besides this, make sure that you set up the correct IP address at the plugin settings, and your printer is available by this IP at the local network (by using something like tracert or ping command )

Jeredian, 2021-08-17:
**closed due to inactivity**

closed 2021-08-17 by Jeredian