PrintMakerLab/mks-wifi-plugin #150: [BUG] Support for new Cura 4.9

BaltasarParreira, 2021-04-20:
Need changes on plugin.json to support new Cura 4.9.

Just change line 6 to:
`    "maximum_cura_version": "4.9",`

vnishukov, 2021-04-21:
After manually change maximum_cura_version, got this error when trying to close cura

[image]

BaltasarParreira, 2021-04-21:
Yes I have also the error, but apart that the plugin works normally.
Going to investigate what else needs to change to avoid that.

Xoxe2, 2021-04-21:
After changing the text line the plugging worked.
When closing the application either it closes unexpectedly or it gives me the following error.
[error.pdf](https://github.com/Jeredian/mks-wifi-plugin/files/6352212/error.pdf)

geruvenier, 2021-04-22:
After manually change maximum_cura_version, got this error when trying to copen cura, and close the program. The same error got on Cura 4.8.

Traceback (most recent call last):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\mks-wifi-plugin-develop\MachineConfig.py", line 121, in restartDiscovery
    self._network_plugin.startDiscovery()
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\mks-wifi-plugin-develop\MKSOutputDevicePlugin.py", line 90, in startDiscovery
    self._zero_conf = Zeroconf()
  File "zeroconf\__init__.py", line 2087, in __init__
  File "zeroconf\__init__.py", line 2021, in create_sockets
  File "zeroconf\__init__.py", line 1971, in add_multicast_member
OSError: [WinError 10022] Se ha proporcionado un argumento no válido

Thread 0x00002f34 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000d14 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000254c (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\mks-wifi-plugin-develop\MKSOutputDevicePlugin.py", line 273 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000032c0 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002b70 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003d14 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003718 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000984 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002ca4 (most recent call first):
  File "threading.py", line 302 in wait
[... 87 lines cut ...]
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003c88 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00003d18 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

jellespijker, 2021-04-23:
Keep in mind that the `global_container_stack` can return `None`. This typing didn't change in 4.9 (https://github.com/Ultimaker/Uranium/blob/master/UM/Application.py#L99) but there are now situations where we set the `global_container_stack` to `None` this was never a likely scenario before. but because of the necessary upgrade to Qt 5.15.2 we now do set it to None in some cases.  Skimming through the trace log of @vnishukov would suggest it returns a `None`.
I think that if you handle this case gracefully this bug might be resolved.

Jeredian, 2021-04-23:
Duplicate of #148

closed 2021-04-24 by Jeredian

jellespijker, 2021-04-25:
@Jeredian I assume you have uploaded to new plugin to the marketplace. If so I'm willing to take a look today. Otherwise it will be a couple of days before we actually have chance to process this, since there is national holiday on Tuesday in the Netherlands.

Jeredian, 2021-04-25:
@jellespijker , sure, I planned to upload the update in half an hour, I'm waiting last tests result from the community at the 1.2.2 RC1 version. Our milestone is not so prioritized, then supporting new API. 
Thank you for your support and for your time.

BaltasarParreira, 2021-04-25:
Hi, so far looks like is working ok. 
I just git clone now and put on top of the 1.2.1 installed and I can after open Cura plugin is there and I can connect to the printer.
Also on closing Cura the error reported here disappeared.
Going to do tests sending files to print and so on now, let you know later how this tests worked.
Thanks.

Xoxe2, 2021-04-25:
Hi, so far everything seems to be working fine.
I have reinstalled the plugin manually and the shutdown and unexpected shutdown errors appear to have been fixed in my OS X 10.13 High Sierra environment.
Now it only remains to test the stability by printing but it seems that it looks good.
Thanks.

Jeredian, 2021-04-25:
@BaltasarParreira , @Xoxe2 , thanks. 
After #157 will be merged, I'll send the new release to the marketplace.

Jeredian, 2021-04-25:
@jellespijker, I submitted a new release to the marketplace. 
https://github.com/Jeredian/mks-wifi-plugin/releases/tag/1.2.2

jellespijker, 2021-04-25:
The package is approved and ready to be downloaded/upgraded from the marketplace