PrintMakerLab/mks-wifi-plugin #151: the plugin dosen't work with 4.9

msmeeh, 2021-04-21:
the pulgin doesn't work with Cura 4.9, please update it or give solution

JonathanMortes, 2021-04-22:
It doesn't work either with 4.7.1 either. I've got their software and HW (MKS robin e3p and TFT35) and it won't show me the preview on the machine and sometimes it will even fail. All I get is a G on the screen

Jeredian, 2021-04-23:
@JonathanMortes , at now plugin, support Cura from 4.5 to 4.8, so it can be installed from Cura Marketplace and connect to the printer.
About preview, since the latest update, was moved to the plugin settings, you can check "image settings" at the same place where you add IP settings for your printer.

Jeredian, 2021-04-23:
@msmeeh , FYI please check already existed issues instead of making a duplicate issue with so obvious thing
you can look to this #148

Jeredian, 2021-04-23:
Duplicate of #148

closed 2021-04-23 by Jeredian