PrintMakerLab/mks-wifi-plugin #154: [BUG]

Pro-serj, 2021-04-23:
**Describe the bug**
No button for WiFi connect to printer on Cura 4.9.0

**To Reproduce**
Steps to reproduce the behavior:
1. Update Cura from 4.8 to 4.9
2. install MKS WiFi plugin
3. No button for WiFi connect to printer

**Expected behavior**
Connection to printer by Wifi

**Screenshots**
[image]
[image]
[image]

Desktop (please complete the following information):
 - OS:  Windows 10
 - Cura version:  4.9.0
 - Plugin version: 1.2.1
 - 3D printer: FlyingBear Ghost 5
 - Motherboard and/or WiFi module firmware version: V5.8-T25

There is no button to connect to Printer by WiFi
On Cura 4.8.0 it was working, but sometimes if conncected to printer and closed Cura (without disconnection) can't connect after restarting Cura... But after updating Cura to 4.9.0 it does not show any button to connect to printer.

wkoetz, 2021-04-23:
Did you try this workaround?
https://github.com/Jeredian/mks-wifi-plugin/issues/148#issuecomment-822040166

Pro-serj, 2021-04-24:
Yes I did, but even the folder name isn't same as described in Mac...

wkoetz, 2021-04-24:
I have done this patch. It works well.

Jeredian, 2021-04-24:
@Pro-serj , FYI please check already existed issues instead of making a duplicate issue.

About file folder, RTFM . 
• Windows: %APPDATA%\cura\ , or C:\Users\AppData\Roaming\cura\
• OSX: $USER/Library/Application Support/cura/
• Linux: $USER/.local/share/cura/
inside that folder, you need to go inside "cura-version" -> "plugins folder" -> "mks wifi plugin folder".

Jeredian, 2021-04-24:
Duplicates #148 , #150

Jeredian, 2021-04-24:
Duplicates of #148

closed 2021-04-24 by Jeredian