PrintMakerLab/mks-wifi-plugin #166: [FEATURE] SD card multiple files deletion

firstvadim, 2021-04-27:
It's quite annoying to delete files one by one with select file window closing every time.
So please, make multiple file selection possible in SD files window. The print button should be grayed out in case of more than one file selected. Or it could be two different windows - one for printing from card and another one for cleaning.

zmp2000, 2021-06-13:
File deletion for me is completly not working, I hit the delete button but nothing happens, I'm currently using 1.2.2 version, is that a bug or Am I missing something?

Elkin-Vasily, 2021-11-15:
> File deletion for me is completly not working, I hit the delete button but nothing happens, I'm currently using 1.2.2 version, is that a bug or Am I missing something?

Could you please open a new bug issue with detailed description of the problem?