PrintMakerLab/mks-wifi-plugin #167: Add webcam function

aperoap1, 2021-05-08:
That would be great if you could intergrien webcam With a link like "http://ip/webcam/?action=stream"

Nexurion, 2021-11-24:
And it would be great if you could integrate a webcam via the USB. For example, I have a Logitech C920 connected to my computer and I have to monitor printing through OBS Studio. It would be convenient if the camera would be displayed in the "MONITOR" tab in Cura.

Nexurion, 2021-12-02:
@Jeredian, thanks for the answer! I will follow the updates and wait for the appearance of this feature. I still have little experience in 3D printing and the functionality of my printer is enough for me, so I don't want to changing to OctoPrint. Good luck to you in development and in life! :)

closed 2024-03-12 by Jeredian