PrintMakerLab/mks-wifi-plugin #174: Dont work with cura 4.10.0 [BUG]

TimeJammer, 2021-07-02:
Dont work with cura 4.10.0

Manuel or cura market.

commo81, 2021-07-03:
Same thing here. Installed from marketplace and manually, "cannot communicate with printer at IP address." I've only had it successfully work on Cura 4.2.1 so I'm using that for now.

Jeredian, 2021-07-04:
Duplicate of #171

Jeredian, 2021-07-04:
Please check the already exited issues, before creating a new one.

emare62, 2021-07-07:
I have checked the issue #171but did not understand what to do to fix this.
After some tests I edited file plugin.json :
 ```
old     "maximum_cura_version": "4.9"
new     "maximum_cura_version": "4.10"
```
now seems to work

Jeredian, 2021-07-07:
You also need to change these lines:
```
  "supported_sdk_versions": [
      "7.1.0",
      "7.2.0",
      "7.3.0",
      "7.4.0",
      "7.5.0"
  ]
```
after that, it will be shown, but if there are some breaking changes it would not work and you'll need to check the cura.log file

Jeredian, 2021-07-07:
I'll start working on #171 today, after work. So tomorrow there will be an update in the dev branch.

TimeJammer, 2021-07-07:
Thanks for your answers and hard work. Good luck with.

TimeJammer, 2021-07-07:
> I have checked the issue #171but did not understand what to do to fix this.
> After some tests I edited file plugin.json :
> 
> ```
> old     "maximum_cura_version": "4.9"
> new     "maximum_cura_version": "4.10"
> ```
> 
> now seems to work

Thanx for fix. 
It's seems working for me too...
At least it works for now. Thanks!

PR #176 (2021-07-10): [MWP-171] [FEATURE] Update Cura version
closes #171
closes #173
closes #174
closed 2021-07-10 by Jeredian