PrintMakerLab/mks-wifi-plugin #177: Doesent work with cura 4.10

EgorIII, 2021-07-19:
There is no "mks wifi connection" button on Cura 4.10, but plugin is installed and enabled.
works correctly on older versions 

 - OS: Windows 10
 - Cura version: 4.10
 - Plugin version: 1.2.2
 - 3D printer: FlyingBear Ghost 4S

aperoap1, 2021-07-21:
You can download the current version: [https://github.com/Jeredian/mks-wifi-plugin](https://github.com/Jeredian/mks-wifi-plugin)
Then open cura -> help -> show configuration folder 
(C:\Users\surface3\AppData\Roaming\cura\4.10\plugins\MKSWifiPlugin\MKSWifiPlugin)

Unzip the downloaded zip file and copy everything in there (replace data). Then restart cura.
It worked for me.

diviatrix, 2021-07-28:
Sadly it wont connect
https://i.imgur.com/CfD04R8.png

Jeredian, 2021-07-30:
@diviatrix , it seems that you need to add an IP address of your printer. You can't connect if there is nothing to.

diviatrix, 2021-07-30:
@Jeredian it is actually added, and this is what it looks like :)

Jeredian, 2021-07-30:
@diviatrix, please create a separate issue about it, we will investigate this. Because on my mac and pc it works as it should. 
Have you tried to toggle the wifi support checkbox and add your IP address again?

marsaldev, 2021-07-30:
Same here, no button for configuration:

- OS: macOS Big Sur 11.4
- Cura version: 4.10
- Plugin version: 1.2.2 (latest)
- 3D printer: FlyingBear Ghost 5

[image]

[image]

Jeredian, 2021-07-30:
@marsaldev  , please read comments, and take a look at the pinned issues that are already done, before comment :-)

https://github.com/Jeredian/mks-wifi-plugin/issues/177#issuecomment-884512417

For now, you can use the version from dev. branch, next release will be at the start of august

marsaldev, 2021-07-30:
@Jeredian Thanks a lot

Pumpken, 2021-07-30:
Developer version adds the button but still cant connect. Same as @diviatrix. But its nice to know that the DEV is on it.

Jeredian, 2021-07-30:
@Pumpken, @diviatrix can you tell, please, your os version. maybe I will try to fix it this weekend. And it will be great if someones can show a gif or video (screencast) that describes your actions because we can fix only what we can reproduce.

Jeredian, 2021-07-30:
@Pumpken , @diviatrix , btw after update to the current develop version, you need to set your settings like it's the first time.
So it looks like a conflict of version with previous settings. You can try to clear the plugin folder and copy it from develop, maybe it can fix this. (because when Cura updates a plugin it does the same thing)

Jeredian, 2021-08-06:
closed due inactivity

closed 2021-08-06 by Jeredian