PrintMakerLab/mks-wifi-plugin #178: [BUG][1.2.2] Set point temperature of extruder 2 not set to 0 after cancel

bellzw, 2021-07-24:
**Describe the bug**
With BIBO 2 Touch printer, when Cancel is clicked, set point of extruder 2 is not set to 0, even though heater 2 is actually off.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect BIBO 2 printer
2. Set a preheat temperature and click Pre-heat under Extruder 2
3. Wait for set point (or not, if you're impatient)
3. Click Cancel under Extruder 2
4. Pre-heat button reappears, but previous set point remains to the right of the current temperature

**Expected behavior**
Set point should be set to 0 indicating heater is off

**Screenshots**
[image]
The issue is highlighted in yellow.  Heater is actually off and temperature is at room temperature, but screen indicates the set point is 160 C.  Extruder 2 is ready to be heated to 200 C.  This does not happen on the Extruder 1 side, where 0 C is shown and the heater is really off.

**Desktop (please complete the following information):**
 - OS: Windows 10 64-bit on Lenovo Yoga 14 laptop
 - Cura version: 4.9.1
 - Plugin version: 1.2.2
 - 3D printer: BIBO 2 dual touch
 - Motherboard and/or WiFi module firmware version: 3.03

**Additional context**

Add any other context about the problem here.

Elkin-Vasily, 2021-11-01:
Can you attach Cura log file?

Elkin-Vasily, 2021-11-28:
I have tested this. Plugin works just correct (it sends the needed command M104 S0 T1), there is some issue in the wifi module or the printer firmware. Try to make an issue here: https://github.com/makerbase-mks/MKS-WIFI

closed 2021-11-28 by Elkin-Vasily

bellzw, 2021-11-28:
I have been traveling and will be check this again after I return home on 29 Nov.  Note that the problem was NOT that the printer nozzle stayed hot, it was that MKS WiFi continued to show the previous set point.  Since posting this issue, I have updated Cura to 4.12, so I don't yet know if the problem persists.

Elkin-Vasily, 2021-11-28:
I have tested this with Cura 4.12.1 and MKS WiFi Plugin 1.2.7-dev. As I sad the problem is not in plugin. Plugin sends the right command to set target temperature to 0, but WiFi-module continue to send that target temperature for the second extruder is the same as it was (no matter what it was 200 or 1 degree Celsius).

bellzw, 2021-11-29:
Understood. I'm at the airport as I write this and will check it on my installation later today or tomorrow.  I just did the update to 4.12.1 and updated the plug-in.  If indeed the printer reports the wrong temperature back to Cura, I'll notify the manufacturer that there's a bug in their firmware.

bellzw, 2021-12-01:
I don't seem to have access to 1.2.7-dev from Cura; I can only get 1.2.6, which I have installed.  Do I need to get 1.2.7-dev from github and manually install it?

There appears to be some problems with 1.2.6 as follows:

1. I get this screen:
[image]
  The IP address is almost illegible with grey against a black background.  It would be better with black on white background.
  Both check boxes need to be checked before Maximum file name length, Add, Edit, or Remove become active.  Why does the user need to check the boxes since the plug-in is installed for the purpose of letting Cura control the printer?  Seems like it would make more sense to have them both checked by default.
  Remove doesn't work.  I can select the address, click remove, and the address remains.  It even remains if I uninstall 1.2.6, close Cura (4.12.1), restart Cura, and reinstall 1.2.6.
  Under Preview Settings, there are only a few printer models, none of which are the installed printer.
  I don't know what Simage and Gimage mean.  These are for screen shots?  Taken by whom and of what screen? Is there a help file somewhere?

2.  1.2.6 does connect to the printer.  However, when I look at the Monitor screen, the model (not transferred to the printer, yet) is not shown. 
[image]
Does the Monitor screen only show what the printer is doing, and it shows nothing until it's actually printing?

I've attached the Cura log file.
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7630608/cura.log)
I see that the printer appears to be reporting T1: nn / 40 even after the M104 S0 T1 is sent, which would indicate a problem with the printer's firmware.  

What is the purpose of M997?

I have Pronterface installed on my computer.  So, I removed the SD card, and connected a USB cable between the computer and printer, and connected to it with Pronterface.  Then, I sent M105 commands to see how the printer responded, followed by M104 to change the temperature of E1, followed by M105 to see what was reported, and finally I sent M104 T1 S0 and and M105.  You'll see the stream at the right in the screenshot below.
[image]
You'll see that via the USB port on this printer, after I turn off the heater, the printer reports T1 temp/0, not /40.  I don't know if it is reasonable to think the printer would respond differently via USB and WiFi, but at face value, that seems to be the case.  I'll check with the manufacturer about this.

Elkin-Vasily, 2021-12-02:
> Do I need to get 1.2.7-dev from github and manually install it?

Actually no. There some minor changes that have not been tested properly yet.

> The IP address is almost illegible with grey against a black background.  It would be better with black on white background.

We are getting this elements from Cura Theme. You can change Cura's current theme, it can influence this.

> Both check boxes need to be checked before Maximum file name length, Add, Edit, or Remove become active.  Why does the user need to check the boxes since the plug-in is installed for the purpose of [...]

This made for those who has multiply printers and not all of them with MKS WiFi-module or MKS TFT.

>    Remove doesn't work.  I can select the address, click remove, and the address remains.  It even remains if I uninstall 1.2.6, close Cura (4.12.1), restart Cura, and reinstall 1.2.6.

Remove works only when MKS WiFi Plugin is on for active printer (with a checkbox). 

>    Under Preview Settings, there are only a few printer models, none of which are the installed printer.

Just use "Default" option (mostly it is that you need), otherwise you can create issue in this repo with request of adding your printer type.

>    I don't know what Simage and Gimage mean.  These are for screen shots?  Taken by whom and of what screen? Is there a help file somewhere?

This parameter needed for generate preview image for different printers. They can be modified only when "Custom" option is selected (for the most printers "Default" will be enough).

>    Does the Monitor screen only show what the printer is doing, and it shows nothing until it's actually printing?

We are using Cura form for this tab (with some little additions), so answer is yes.

> What is the purpose of M997?

You can check the purpose of MKS WiFi module commands by reading this [file](https://github.com/makerbase-mks/MKS-WIFI/blob/master/Transfer%20protocal%20%20between%20ESP%20and%20MCU.docx).

>  I don't know if it is reasonable to think the printer would respond differently via USB and WiFi, but at face value, that seems to be the case.

Via USB answers the main-board directly. WiFi module translates the commands to the main-board, but not all of them, for some commands module answers by him self, and some main-board answers are changing by WiFi module.