PrintMakerLab/mks-wifi-plugin #196: [BUG] MKS WiFI 1.2.4 won't install on Cura 4.10 (Manjaro Linux)

ilfarpro, 2021-08-16:
MKS WiFI Plugin won't install. 

**To Reproduce**
Steps to reproduce the behavior:
1. Download or clone latest release;
2. Open Cura configuration folder. It can be found via Help -> Show Configuration Folder inside Cura;
3. Copy folder into [Cura configuration folder]/plugins/ ;
4. Restart Cura.

Nothing happens at all. If open Store, Installed Plugins tab it will show nothing. Cura just doesn't see plugin. 

If I could provide more information on that, please, tell me how, I want to help.

**Desktop (please complete the following information):**
 - OS: Manjaro Linux KDE (latest)
 - Cura version: 4.10
 - Plugin version: 1.2.4
 - 3D printer: FlyingBear Ghost 5
 - Motherboard and/or WiFi module firmware version: V5.8/V5.8

Jeredian, 2021-08-17:
@ilfarpro, It seems that you missed something when updated the plugin manually.
Let's do it as a more simplest way:
1. Install plugin from Cura marketplace 
2. Go to the plugins folder and find `MKSWifiPlugin` folder
3. Go inside it and found `caches` folder (it will be named like that: '`__pycache__`'). Remove it
4. Replace all files with new ones
5. Restart Cura
6. If something goes wrong, attach the Cura log ( $USER/.local/share/cura//cura.log )

ilfarpro, 2021-08-17:
Thank for reply!

I followed your steps few times, and it did not helped. Plugin still not working at all. 
But I found errors in cura.log file. Pay attention to line 417. It starts from there. 

I think now it's not just question, it should be a bug...

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/6999825/cura.log)

Jeredian, 2021-08-17:
As I saw inside the log file, this issue repeats also for  `UM3NetworkPrinting`, the main issue is that Cura cannot find one of the modules: `ModuleNotFoundError: No module named 'zeroconf'`.
And for some reason it trying to search in here :
`File "/usr/lib/python3.9/site-packages/UM/PluginRegistry.py", line 681, in _findPlugin`.

So as you can see this issue is not related to our plugin, it's mostly about Cura and zeroconf module.
You can try to install `zeroconf` manually, and it should fix your problem

ilfarpro, 2021-08-17:
@Jeredian Thank you so much! It worked. Is there any solution to prevent other users from such a problem? Maybe I should open issue in Cura repository?

Jeredian, 2021-08-17:
Sure, you definitely can create an issue at Cura's repository. I think they will have more information about common solvation. Or maybe it's just a random issue with the continuous deployment process.

Btw, you can join our telegram group ( https://t.me/mks_wifi_plugin_dev ) , for some plugin updates news, or questions that can be solved by discussion faster than here.

closed 2021-08-17 by Jeredian