PrintMakerLab/mks-wifi-plugin #201: It doesn't work well with Cura 4.11.0

TimeJammer, 2021-09-09:
It doesn't work well with Cura 4.11.0. First, cura never saw the plugin, then I made the "maximum_cura_version "4.11.0" in the Plugin.json file. In this case, it sees the plugin, but it cannot send to the printer properly. Also, Preview settings are reset from custom to default every time.

Mks Wifi Plugin Version: 1.2.5 dev

Jeredian, 2021-09-10:
@TimeJammer , can you attach a log file, please (cura.log), with the timeline since Cura started and when you changed preview settings?
It can work not well because right now it's a new version of Cura and nobody tested compatibility yet. I think there can be some minor issues, and we can try to solve them asap.

Jeredian, 2021-09-10:
@TimeJammer ,  I checked at the last version of Cura (4.11) and the plugin settings on my PC were correctly saved after I set them up and restart Cura. What type of operating system do you use? Or this part of the issue was about settings reset after each Cura or plugin update?
[image]

Jeredian, 2021-09-10:
Plugin compatibility was fixed in #197, and it's already in the `develop` branch. So `plugin.json`,  will be updating mostly for SDK changes (as it should, because it may break something)

TimeJammer, 2021-09-10:
After Cura is closed and opened, it turns from default to custom. In addition, when we send a file with the same name to the printer, the upload does not start even if we delete the old one with the same name. I reinstalled Cura. I also tried the relaese you just posted. I couldn't find the cura.log.

Every time I open and close Cura;
From this,
[image]
it goes back to this.
[image]

I tried reinstalling Cura. The same problem continues.

Jeredian, 2021-09-10:
@TimeJammer , understood. I will investigate this. It looks like when the popup is open screenshot type is not updated in _combobox_ to the current one, but _simage_ and _gimage_ are correct.
You can find the `cura.log` file by one of this paths:
• Windows: %APPDATA%\cura\cura.log , or C:\Users\AppData\Roaming\cura\cura.log
• OSX: $USER/Library/Application Support/cura//cura.log
• Linux: $USER/.local/share/cura//cura.log

TimeJammer, 2021-09-10:
I've realized now. It is reset not when I close Cura, but when I close the Preferences window. I am uploading the new config file since I deleted and installed Cura. Thanks for the guide.
Log:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7143650/cura.log)

CyberTechX, 2021-09-10:
Try edit plugin.json file and change the line 6:
    "maximum_cura_version": "4.10.0",
to
    "maximum_cura_version": "4.11.0",

emare62, 2021-09-22:
Hi, I have the same issue as TimeJammer (first comment on top).
Cura 4.11.0 does not see MKS WIfi Plugin 1.2.4
I'm running Win10 Enterprise.
I uninstalled completely Cura 4.10 as well as MKS, then installed 4.11 and MKS 1.2.4

Please let me know if you need more info.
Thanks

[image]
[image]

Jeredian, 2021-09-22:
@emare62 , please read a full topic before post a message next time.

As I said here (https://github.com/Jeredian/mks-wifi-plugin/issues/201#issuecomment-916812563) :
`Plugin compatibility was fixed in #197, and it's already in the develop`

You can manually install the latest version from `develop`. Or wait until it will be on the Cura Marketplace

siprem10, 2021-10-02:
> @ emare62 , lea un tema completo antes de publicar un mensaje la próxima vez.
> 
> Como dije aquí ( [# 201 (comentario)](https://github.com/Jeredian/mks-wifi-plugin/issues/201#issuecomment-916812563) ): `Plugin compatibility was fixed in #197, and it's already in the develop`
> 
> Puede instalar manualmente la última versión desde `develop`. O espere hasta que esté en Cura Marketplace

The plugin does not work for me and I downloaded it from github.
I have cura 4.11, the plugin and the MKS WIFI options appear, even after segmenting a .stl file it shows me "Save as TFT file", but when I pass it to the printer the thumbnail is not shown. I tried older versions of the plugin and it works for me without problems.

Google translate xd