PrintMakerLab/mks-wifi-plugin #233: How to add metadata "simage "and "gimage" in GlobalContainerStack for wanhao D12 printer

jef-maubert, 2021-11-18:
### What is your question?

First of all thanks a lot for this plugin, that I do appreciate.

The miniature is no included into the gcode with the "MKS Wifi Plugin" , while it was correctly included into the gcode with the  "MKS Plugin Connection".
Comparing the sources of the 2 plugins, it seems this is due to the way each plugin define the size of the miniature : 

- in "MKS Plugin Connection", those size where respectively hardcoded as 100 and 200 in the "save_gcode" function of the "SaveOutputDevice.py" file. 
- in "MKS Wifi Plugin" the function "add_screenshot()" of "utils.py" search the size of the simage and the gimage in the  "GlobalContainerStack". As it looks like there are no such entry in this "GlobalContainerStack", the miniature is not inserted.

As a workaround, I have modified the "add_screenshot()" function to use a constant value when there is no entry in the stack. And this works fine, but this is ugly. 

Thus my question is : How to insert properly the definition of simage and gimage (as values 100 and 200) into the definition of the Wanhao D12 printer ?

Best regards
Jef Maubert

Jeredian, 2021-11-20:
Hi @jef-maubert , 
We have a file with screenshot settings (https://github.com/Jeredian/mks-wifi-plugin/blob/develop/config/screenshot.json)
You can edit it to use a new value for your Printer model.
But also we have a ticket (#232) to implement it in the next versions of the plugin, probably on 1.2.7

Elkin-Vasily, 2021-11-27:
Have been implemented in current dev-version during #232 issue

closed 2021-11-27 by Elkin-Vasily