PrintMakerLab/mks-wifi-plugin #234: MKS WiFi - CURA 4.12

1lug1998, 2021-11-19:
### Plugin Version

The latest version

### Cura Version

4.12

### Platform

Windows 11

### Printer

FLSun Cube with SKR 2 Mainboard

### Reproduction steps

I'm seeing some issues with the MKS wifi plugin and CURA 4.12.
The file shows that it uploaded successfully both in CURA and on the MKS TFT32 V3.0 I have. I cannot see the file on the screen. If I view the SD card in cura, it does not show the file. When I remove the SD card from the screen and plug it into my PC, I see the file name with no file extension, almost as if it did not upload as .gcode.

Any ideas?

### Actual results

There is not printable file on the SD card. 

### Expected results

I expected a printable .gcode file on the SD card. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have no file to upload unfortunately since there is no file extension.

Elkin-Vasily, 2021-11-19:
Attach a log-file, please

1lug1998, 2021-11-19:
> 

I'm not quite sure how to get a log file of the issue.

1lug1998, 2021-11-19:
> Attach a log-file, please

Log file has been uploaded.

Elkin-Vasily, 2021-11-20:
In the log file you have send I see that you try to upload 20mm_cube.gcode file, it works ok. And then I see this file in file list on your SD:

mks recv: Begin file list
mks recv: System Volume Information.DIR
mks recv: four_square_cons.gcode
mks recv: 4132-M-052REVB.gcode
mks recv: 20mm_cube.gcode
mks recv: End file list

Was there some problems with this file?

1lug1998, 2021-11-22:
> In the log file you have send I see that you try to upload 20mm_cube.gcode file, it works ok. And then I see this file in file list on your SD:
>
> mks recv: Begin file list mks recv: System Volume [...]

20mm_cube uploaded just fine weirdly, I haven't had a good upload in a while. The others I put on the SD card by inserting the SD card into my PC to transfer them.

closed 2021-11-26 by 1lug1998