PrintMakerLab/mks-wifi-plugin #239: MKS WiFi Plugin wont connect to ESP8266 flashed with MKS-WiFi firmware

Reasonable-Human, 2021-11-25:
### Plugin Version

1.2.2

### Cura Version

4.12

### Platform

Windows 11

### Printer

A custom Marlin Printer, connected to an esp8266

### Reproduction steps

Ive installed the plugin from the marketplace in Cura 4.12. Then when i open the MKS WiFi Plugin window on my printer in the printers section i enable the WiFi and add the IP address that my esp8266 (which was flashed with [this](https://github.com/makerbase-mks/MKS-WIFI)). I then hit the connect button at the bottom but nothing happens. And if i exit out of the menu and re-enter it the WiFi support will have toggled itself off.

### Actual results

Nothing, the monitor section still asks for me to connect the printer, i can still save as a TFT file but that seems to always be available.

### Expected results

I would expect the monitor window to show information about the printer and for some way to print from the slicing window.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have the ESP8266 connected over the aux port and replacing the firmware with ESP3d lets me interact with the printer. I can also interact with the web interface of the esp with no issues

Jeredian, 2021-11-26:
@Reasonable-Human , please attach a log file.

Reasonable-Human, 2021-11-26:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7609462/cura.log)
Here it is.

Elkin-Vasily, 2021-11-27:
As I see in log-file plugin sends commands to wifi-module, but gets no answers.

You cat try ti connect to wifi-module by using netcat or telnet with printer IP and 8080 port. The connection can be tested with M105 and M997 commands.

Reasonable-Human, 2021-11-27:
If i telnet to it with the command `telnet 192.168.178.134 8080` the terminal goes blank. However nothing i type shows up in the terminal, and typed commands (like m80, m105) have no response.

Elkin-Vasily, 2021-11-28:
It means that you have some problems with wifi-module, or with the printer firmware.

closed 2021-11-29 by Jeredian